Affirmatech/MeshSense #1: Text messages via the app fail to send properly if the connected node has no physical radio

thespillmonkey, 2024-09-16:
In order to not tie up the TCP API on my base station node, I run quite a few virtual nodes. 

These are VMs of debain12 running meshtasticd. All virtual nodes connect to a local MQTT broker that passes all traffic from the main base station node. In this way, each virtual node is a fully functioning node on the mesh and has it's own configuration. 

I can then connect things like my BBS, my ChatGPT bot, or MeshSense to a dedicated virtual node, and they simply talk over MQTT instead of a radio, and then the base station is their first hop out. It works perfectly. 

Meshsense is scraping data correctly over it's virtual node, and initiate's it's traceroutes properly. 
When I try to send a message out however, it throws an error and isnt sent out. Possibly there is some unique behavior when it senses the "Simulator_App" type? Not sure. 

[image]

Soltares, 2024-09-16:
Interesting use case!  Are you able to please share the contents of the magnifying glass 🔎 for both SIMULATOR_APP and the undefined line underneath.  I hadn't encountered that packet type during development.  Thanks!

thespillmonkey, 2024-09-17:
reach out to me on the meshtastic discord server - Gary - WG4RY [E][VE]
I will give you access to the web interface. I have it public facing.

thespillmonkey, 2024-09-17:
{
  "from": 1681692781,
  "to": 4294967295,
  "channel": 0,
  "decoded": {
    "portnum": "SIMULATOR_APP",
    "payload": "CAESFlRlc3QgU2VuZCBvZiBUZXh0IERhdGE=",
    "wantResponse": false,
    "dest": 0,
    "source": 0,
    "requestId": 0,
    "replyId": 0,
    "emoji": 0
  },
  "id": 705534539,
  "rxTime": 1726537019,
  "rxSnr": 0,
  "hopLimit": 0,
  "wantAck": false,
  "priority": "DEFAULT",
  "rxRssi": 0,
  "delayed": "NO_DELAY",
  "viaMqtt": false,
  "hopStart": 0,
  "publicKey": "",
  "pkiEncrypted": false
}

AND

{
  "event": "onSimulatorPacket",
  "json": {
    "id": 705534539,
    "rxTime": "2024-09-17T01:36:59.000Z",
    "type": "broadcast",
    "from": 1681692781,
    "to": 4294967295,
    "channel": 0,
    "data": {
      "0": 8,
      "1": 1,
      "2": 18,
      "3": 22,
      "4": 84,
      "5": 101,
      "6": 115,
      "7": 116,
      "8": 32,
      "9": 83,
      "10": 101,
      "11": 110,
      "12": 100,
      "13": 32,
      "14": 111,
      "15": 102,
      "16": 32,
      "17": 84,
      "18": 101,
      "19": 120,
      "20": 116,
      "21": 32,
      "22": 68,
      "23": 97,
      "24": 116,
      "25": 97
    }
  }
}

Soltares, 2024-09-18:
I added code with the intent of decoding the SIMULATOR_APP message and displaying it in the log.  Regarding sending messages via MQTT, it's not entirely clear.

MeshSense uses the following API call to send a message:
https://js.meshtastic.org/classes/HttpConnection.html#sendText

It takes a destination node number or channel id when sending.  (I don't think it worked quite right if I provided both values)

I also came across this while researching:
https://meshtastic.discourse.group/t/what-is-the-correct-way-to-send-to-mesh-from-mqtt-now/11301

> the established way to send to the mesh doesn’t work unless the channel is named “mqtt”

Perhaps it needs to send over an explicit "mqtt" channel?  I'm not set up to test this at the moment, but I've published `1.0.14-beta.2` that has the SIMULATOR_APP decode code implemented.  If you enable "beta updates" in settings it should download this version.