Affirmatech/MeshSense #16: App Crash on M1 Mac Mini

TheAZtech, 2024-09-25:
App crashes after a few moments. Included is the Console file for the crash report. 
[MeshSence Helper.txt](https://github.com/user-attachments/files/17123053/MeshSence.Helper.txt)

Soltares, 2024-09-27:
Thank you for the report!  We have had similar reports regarding bluetooth on Mac systems can cause problems and that disabling the adapter can get the application to run.  I found that the simpleble library will throw an unhandled exception inside a setTimeout if it is unable to initiate a bluetooth scan causing the application to crash.

I am considering implementing a catch-all `process.on('unhandledException')` since this problem is seemingly inside 3rd party library code, but I know this isn't the greatest practice.  A try/catch block won't work as the exception occurs in a setTimeout call from `webbluetooth`.

Soltares, 2024-10-29:
@TheAZtech Can you please confirm if this crash continues to occur in the latest MeshSense version (1.0.15)

Some code was added to prevent crashing particularly when the OS doesn't want to share access to the bluetooth adapter.

TheAZtech, 2024-10-31:
@Soltares Thank you for looking into this. I updated the app and I am still seeing it crash anytime anything is touched here. (took me three tries to get this screen shot) 
[image]
it is also sporadically crashing if just left alone, I am able to click just about everywhere with no issue, until I touch this area.

recrof, 2024-11-26:
Newest version, macbook air m1 - headless mode crash(GUI mode crashes the same):
```
/Applications/MeshSense.app/Contents/MacOS ⌚ 21:54:23
$ HEADLESS=1 ./MeshSense
DIRNAME /Applications/MeshSense.app/Contents/Resources/app.asar/out/main
API_PATH /Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [ '/Applications/MeshSense.app/Contents/MacOS/MeshSense' ]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api',
  staticDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/static',
  dataDirectory: '/Users/forcer/Library/Application Support/meshsense'
}
[State] version init
[State] updateChannel init undefined
[State] updateStatus init {}
[State] address init
[State] connectionStatus init disconnected
[State] lastFromRadio init undefined
[State] channels init []
[State] packets init []
[State] nodes init []
[State] currentTime init 1732654475371
[State] myNodeNum init undefined
[State] myNodeMetadata init undefined
[State] accessKey init undefined
[State] packetLimit init 500
[State] apiHostname init undefined
[State] apiPort init undefined
[State] messagePrefix init undefined
[State] messageSuffix init undefined
[State] allowRemoteMessaging init false
[State] autoConnectOnStartup init true
[State] enableTLS init false
[State] automaticTraceroutes init true
[State] meshSenseNewsDate loaded 1732644289102
[State] pendingTraceroutes init []
[State] tracerouteRateLimit init 15
[State] nodeInactiveTimer init 60
[State] bluetoothDeviceList init []
[bluetooth] Begin Scanning
[bluetooth] Adapter available: true
[bluetooth] scanning...
[80731:1126/215435.491442:ERROR:trust_store_mac.cc(821)] Error parsing certificate:
ERROR: Failed parsing extensions

[80731:1126/215435.491486:ERROR:trust_store_mac.cc(821)] Error parsing certificate:
ERROR: Failed parsing certificate policies

Found version 1.0.17 (url: meshsense-arm64.zip, meshsense-arm64.dmg, meshsense-arm64.dmg, meshsense-arm64.dmg)
Downloading update from meshsense-arm64.zip, meshsense-arm64.dmg, meshsense-arm64.dmg, meshsense-arm64.dmg
Checking for macOS Rosetta environment
Checked for macOS Rosetta environment (isRosetta=false)
Checking for arm64 in uname
Checked 'uname -a': arm64=true
updater cache dir: /Users/forcer/Library/Caches/meshsense-updater
Unable to locate previous update.zip for differential download (is this first install?), falling back to full download
[api to electron] { event: 'setUpdateChannel', body: undefined }
[electron] Set update channel undefined
[news] Checking for news
Server listening { address: '::', family: 'IPv6', port: 5920 }
[electron to api] { data: { event: 'version', body: '1.0.16' }, ports: [] }
[State] version set 1.0.16
[electron to api] {
  data: { event: 'checking-for-update', body: 'Checking for update' },
  ports: []
}
[electron to api] {
  data: {
[... 100 lines cut ...]
  data: {
    event: 'download-progress',
    body: {
      total: 103205550,
      delta: 28368977,
      transferred: 85114880,
      percent: 82.4712236890361,
      bytesPerSecond: 20830857
    }
  },
  ports: []
}
[State] updateStatus set {
  event: 'download-progress',
  body: {
    total: 103205550,
    delta: 28368977,
    transferred: 85114880,
    percent: 82.4712236890361,
    bytesPerSecond: 20830857
  }
}
[bluetooth] scanning...
API PROCESS EXITED! 6
```

Soltares, 2024-11-26:
That's strange as initially it seems like everything is loading as it should.  If you disable bluetooth, will the application remain running?  I am able to start it locally here on Apple M2 Sonoma with bluetooth enabled.

recrof, 2024-11-26:
> That's strange as initially it seems like everything is loading as it should. If you disable bluetooth, will the application remain running? I am able to start it locally here on Apple M2 Sonoma with bluetooth enabled.

yes, the app runs as it should. seems it's related how bluetooth stack handles paired devices on mac. if the device is paired on the mac side, but node has been since wiped, it crashes. if you delete all paired nodes that are saved as paired but aren't really it starts to work fine.

Soltares, 2024-11-26:
Thanks for the report!  The only thing I save on the MeshSense side is a string containing the bluetooth device id/address.  When the app launches the `webbluetooth` library will scan for available devices until it finds one with the right id and then makes a connection.

On my mac mini I tried similarly removing my nodes from the Mac bluetooth device list, but for some reason the Mac is not detecting either my Lilygo or WisBlock to re-add :thinking:

Soltares, 2024-11-26:
I tried from MeshSense again and it saw the devices and I was able to connect and re-enter the pin

[image]

[image]

recrof, 2024-11-26:
> I tried from MeshSense again and it saw the devices and I was able to connect and re-enter the pin

you need to reset the node, so mac remembers the pairing, but device doesn't.

jwarchol, 2024-12-06:
I'm having bluetooth related crashes too on Mac. If I turn off bluetooth, Meshsense starts and doesn't seem to crash. If I have bluetooth on, it crashes, sometimes before the UI shows, sometimes a little bit after. I do not have Meshsense configured to auto-connect, so it's something in the discovery side of things. I'm running the latest 1.0.17 version on Sequoia 15.1.1

jwarchol, 2024-12-06:
Example output if I run it headless (I don't typically, but this seems to at least give some output):

```
/Applications/MeshSense.app/Contents/MacOS/MeshSense --headless
DIRNAME /Applications/MeshSense.app/Contents/Resources/app.asar/out/main
API_PATH /Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [
  '/Applications/MeshSense.app/Contents/MacOS/MeshSense',
  '--headless'
]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api',
  staticDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/static',
  dataDirectory: '/Users/joshua/Library/Application Support/meshsense'
}
[State] version init 
[State] updateChannel loaded beta
[State] updateStatus init {}
[State] address loaded 4165C96E-EBE4-55CD-915B-04B20C114BEA
[State] connectionStatus init disconnected
[State] lastFromRadio init undefined
[State] channels init []
[State] packets init []
[State] nodes init []
[State] currentTime init 1733497568636
[State] myNodeNum init undefined
[State] myNodeMetadata init undefined
[State] accessKey init undefined
[State] packetLimit loaded 5000
[State] apiHostname init undefined
[State] apiPort init undefined
[State] messagePrefix init undefined
[State] messageSuffix init undefined
[State] allowRemoteMessaging init false
[State] autoConnectOnStartup loaded false
[State] enableTLS loaded false
[State] automaticTraceroutes loaded true
[State] meshSenseNewsDate loaded 1732644289102
[State] pendingTraceroutes init []
[State] tracerouteRateLimit init 15
[State] nodeInactiveTimer init 60
[State] bluetoothDeviceList init []
[bluetooth] Begin Scanning
Update for version 1.0.17 is not available (latest version: 1.0.17, downgrade is disallowed).
[bluetooth] Adapter available: true
[bluetooth] scanning...
[api to electron] { event: 'setUpdateChannel', body: 'beta' }
[electron] Set update channel beta
[news] Checking for news
Server listening { address: '::', family: 'IPv6', port: 5920 }
CREATING WINDOW
[electron to api] { data: { event: 'version', body: '1.0.17' }, ports: [] }
[State] version set 1.0.17
[electron to api] {
  data: { event: 'checking-for-update', body: 'Checking for update' },
  ports: []
}
[electron to api] {
  data: {
    event: 'update-not-available',
    body: {
      version: '1.0.17',
      files: [Array],
      path: 'meshsense-arm64.zip',
      sha512: 'l/8Xr+s3Brn+GUYqQqzhtqjPwLm6PmY4TZ2aXEd36ZOdH7ioCsIzn+yclXMQ1/1wLfbcURL+VEOFE9VZXtMHaw==',
      releaseDate: '2024-11-26T18:04:23.177Z'
    }
  },
  ports: []
}
[bluetooth] Device Detected | DBF4F286-F8C5-7DA8-A6C4-26391437EEFF SmartShunt HQ2048V9C5S
[WSS] Connection from ::1:52501
[11278:1206/100608.947109:ERROR:vt_video_encode_accelerator_mac.mm(970)] Call NotifyErrorStatus(): code=7, message=VTCompressionSessionCreate failed: Error Domain=NSOSStatusErrorDomain Code=-12908 "(null)"
[11278:1206/100608.947971:ERROR:vt_video_encode_accelerator_mac.mm(970)] Call NotifyErrorStatus(): code=7, message=VTCompressionSessionCreate failed: Error Domain=NSOSStatusErrorDomain Code=-12908 "(null)"
[bluetooth] scanning...
[bluetooth] Device Detected | BF1CC466-4705-6E3F-5EE2-22C2C866D8DA Unknown or Unsupported Device (BF1CC466-4705-6E3F-5EE2-22C2C866D8DA)
[bluetooth] scanning...
[bluetooth] Device Detected | FE14F002-30F5-5984-C2C2-F1502D187FE5 QUEC_BLE_3e92
[bluetooth] scanning...
[bluetooth] Device Detected | FE14F002-30F5-5984-C2C2-F1502D187FE5 QUEC_BLE_3e92
[bluetooth] scanning...
[bluetooth] Device Detected | 30EACA93-0197-B6DC-180B-E3D9C9178452 EF-D3AV3716
[bluetooth] scanning...
[bluetooth] Device Detected | 646C454C-5BA5-0E92-3B02-2D6B9DAC9CFF S1654059728b460e6C
[bluetooth] scanning...
[bluetooth] Device Detected | FE14F002-30F5-5984-C2C2-F1502D187FE5 QUEC_BLE_3e92
API PROCESS EXITED! 6
```

Sometimes there are no devices detected before it crashes, sometimes there's a few more, usually they're in different orders, pretty random which shows up.

Soltares, 2024-12-06:
Thank you @jwarchol for the report and for the log.  I started up MeshSense on my mac mini and let is sit scanning for devices while I went to lunch.  After I came back mine had also seemingly closed/crashed.  At least I am able to reproduce the issue.  It would be nice if finding a solution was as easy :smile:

jwarchol, 2024-12-06:
Please let me know if I can provide any other assistance! Thank you for investigating and thank you for MeshSense!

recrof, 2024-12-13:
Here is stack trace of the bug. it's a problem of `simpleble` library that `npm:webbluetooth` uses.
```
$ DEV_UI_URL=http://localhost:5921 PORT=5920 lldb -- node  --experimental-strip-types src/index.ts
(lldb) target create "node"
Current executable set to '/opt/homebrew/bin/node' (arm64).
(lldb) settings set -- target.run-args  "--experimental-strip-types" "src/index.ts"
(lldb) r
Process 59455 launched: '/opt/homebrew/bin/node' (arm64)
(node:59455) ExperimentalWarning: Type Stripping is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
(node:59455) [MODULE_TYPELESS_PACKAGE_JSON] Warning: Module type of file:///Users/forcer/Downloads/MeshSense/api/src/index.ts is not specified and it doesn't parse as CommonJS.
Reparsing as ES module because module syntax was detected. This incurs a performance overhead.
To eliminate this warning, add "type": "module" to /Users/forcer/Downloads/MeshSense/api/package.json.
{
  programDirectory: '/Users/forcer/Downloads/MeshSense/api',
  staticDirectory: '/Users/forcer/Downloads/MeshSense/api/static',
  dataDirectory: '/Users/forcer/Library/Application Support/meshsense'
}
[State] version init
[State] updateChannel init undefined
[State] updateStatus init {}
[State] address loaded A15273C6-1D7C-0C32-F5A3-DA3A8FE9B601
[State] connectionStatus init disconnected
[State] lastFromRadio init undefined
[State] channels init []
[State] packets init []
[State] nodes init []
[State] currentTime init 1734125961890
[State] myNodeNum init undefined
[State] myNodeMetadata init undefined
[State] accessKey init undefined
[State] packetLimit init 500
[State] apiHostname init undefined
[State] apiPort init undefined
[State] messagePrefix init undefined
[State] messageSuffix init undefined
[State] allowRemoteMessaging init false
[State] autoConnectOnStartup init true
[State] enableTLS init false
[State] automaticTraceroutes init true
[State] meshSenseNewsDate loaded 1732644289102
[State] pendingTraceroutes init []
[State] meshMapForwarding init false
[State] tracerouteRateLimit init 15
[State] nodeInactiveTimer init 60
[State] bluetoothDeviceList init []
[bluetooth] Begin Scanning
2024-12-13 22:39:24.523843+0100 node[59455:3855443] [CoreBluetooth] XPC connection invalid
[bluetooth] Adapter available: true
[bluetooth] scanning...
2024-12-13 22:39:24.547726+0100 node[59455:3855444] [CoreBluetooth] XPC connection invalid
2024-12-13 22:39:24.560556+0100 node[59455:3855443] [CoreBluetooth] XPC connection invalid
2024-12-13 22:39:24.586257+0100 node[59455:3855444] [CoreBluetooth] XPC connection invalid
[news] Checking for news
[meshtastic] Calling connect A15273C6-1D7C-0C32-F5A3-DA3A8FE9B601
Disconnecting from device
21:39:24:604	DEBUG	[iMeshDevice:HttpConnection]	Constructor 🔷 BleConnection instantiated
Already Scanning
[State] connectionStatus set searching
Already Scanning
Server listening { address: '::', family: 'IPv6', port: 5920 }
Process 59455 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x0000000187e7a600 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`__pthread_kill:
->  0x187e7a600 <+8>:  b.lo   0x187e7a620    ; <+40>
    0x187e7a604 <+12>: pacibsp
    0x187e7a608 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x187e7a60c <+20>: mov    x29, sp
Target 0: (node) stopped.
[... 9 lines cut ...]
    frame #7: 0x0000000100262410 node`v8impl::(anonymous namespace)::FunctionCallbackWrapper::Invoke(v8::FunctionCallbackInfo<v8::Value> const&) + 84
    frame #8: 0x0000000100016978 node`Builtins_CallApiCallbackGeneric + 152
    frame #9: 0x0000000100022f68 node`Builtins_LoadIC_NoFeedback + 3912
    frame #10: 0x00000001001833d8 node`Builtins_GetNamedPropertyHandler + 4632
    frame #11: 0x0000000100014838 node`Builtins_InterpreterEntryTrampoline + 280
    frame #12: 0x0000000100014838 node`Builtins_InterpreterEntryTrampoline + 280
    frame #13: 0x000000010001250c node`Builtins_JSEntryTrampoline + 172
    frame #14: 0x00000001000121b0 node`Builtins_JSEntry + 176
    frame #15: 0x00000001005887c0 node`v8::internal::(anonymous namespace)::Invoke(v8::internal::Isolate*, v8::internal::(anonymous namespace)::InvokeParams const&) + 1572
    frame #16: 0x0000000100588120 node`v8::internal::Execution::Call(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, int, v8::internal::Handle<v8::internal::Object>*) + 92
    frame #17: 0x0000000100478bb8 node`v8::Function::Call(v8::Isolate*, v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*) + 176
    frame #18: 0x0000000100266f44 node`napi_call_function + 380
    frame #19: 0x0000000105a46560 simpleble.node`___lldb_unnamed_symbol434 + 68
    frame #20: 0x0000000105a46a80 simpleble.node`___lldb_unnamed_symbol469 + 120
    frame #21: 0x0000000105a46478 simpleble.node`Napi::ThreadSafeFunction::CallJS(napi_env__*, napi_value__*, void*, void*) + 84
    frame #22: 0x00000001002803f4 node`v8impl::(anonymous namespace)::ThreadSafeFunction::AsyncCb(uv_async_s*) + 392
    frame #23: 0x0000000104a51cf4 libuv.1.dylib`uv__async_io + 268
    frame #24: 0x0000000104a62114 libuv.1.dylib`uv__io_poll + 1408
    frame #25: 0x0000000104a52184 libuv.1.dylib`uv_run + 268
    frame #26: 0x00000001001ce488 node`node::SpinEventLoopInternal(node::Environment*) + 256
    frame #27: 0x000000010030759c node`node::NodeMainInstance::Run(node::ExitCode*, node::Environment*) + 192
    frame #28: 0x00000001003072f0 node`node::NodeMainInstance::Run() + 124
    frame #29: 0x000000010027ca68 node`node::Start(int, char**) + 476
    frame #30: 0x0000000187b30274 dyld`start + 2840
```

jwarchol, 2024-12-24:
Still crashing constantly at startup for me, even with the newest beta. If I disable bluetooth the app starts fine (though obviously not too useful apart from TCP). Can we get a build with the dependencies @recrof noted freshened? Or with logging that might suggest what else to look at?  Happy to be the guinea pig here since it's so easy for me to trigger this. Maybe my home environment is just so dense with BLE devices it's tripping something up.

jwarchol, 2025-01-10:
Here's an updated output when crashing on a new MacBook Air:

```
joshua@MacBook-Air:~ % /Applications/MeshSense.app/Contents/MacOS/MeshSense --headless
DIRNAME /Applications/MeshSense.app/Contents/Resources/app.asar/out/main
API_PATH /Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [
  '/Applications/MeshSense.app/Contents/MacOS/MeshSense',
  '--headless'
]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api',
  staticDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/static',
  dataDirectory: '/Users/joshua/Library/Application Support/meshsense'
}
[43179:0109/202312.189174:ERROR:trust_store_mac.cc(821)] Error parsing certificate:
ERROR: Failed parsing extensions

[State] version init 
[State] updateChannel loaded beta
[State] updateStatus init {}
[State] address init 
[State] connectionStatus init disconnected
[State] lastFromRadio init undefined
[State] channels init []
[State] packets init []
[State] nodes init []
[State] currentTime init 1736472192199
[State] myNodeNum init undefined
[State] myNodeMetadata init undefined
[State] accessKey init undefined
[State] packetLimit init 500
[State] apiHostname init undefined
[State] apiPort init undefined
[State] messagePrefix init undefined
[State] messageSuffix init undefined
[State] allowRemoteMessaging init false
[State] autoConnectOnStartup init true
[State] enableTLS init false
[State] automaticTraceroutes init true
[State] meshSenseNewsDate loaded 1732644289102
[State] pendingTraceroutes init []
[State] meshMapForwarding init false
[State] tracerouteRateLimit init 15
[State] nodeInactiveTimer init 60
[State] bluetoothDeviceList init []
[bluetooth] Begin Scanning
Update for version 1.0.18-beta.2 is not available (latest version: 1.0.18-beta.2, downgrade is disallowed).
[bluetooth] Adapter available: true
[bluetooth] scanning...
[api to electron] { event: 'setUpdateChannel', body: 'beta' }
[electron] Set update channel beta
[news] Checking for news
Server listening { address: '::', family: 'IPv6', port: 5920 }
CREATING WINDOW
[electron to api] { data: { event: 'version', body: '1.0.18-beta.2' }, ports: [] }
[State] version set 1.0.18-beta.2
[electron to api] {
  data: { event: 'checking-for-update', body: 'Checking for update' },
  ports: []
}
[electron to api] {
  data: {
    event: 'update-not-available',
    body: {
      version: '1.0.18-beta.2',
      files: [Array],
      path: 'meshsense-beta-arm64.zip',
      sha512: '+UCrDB0z8EJPi2r4A/2dybd5YHgTmZ8LXLBknreP0y9DWtRBfl+4YxkwzpbBkBtHMjK5v9+VKlutXiOxOoRJXg==',
      releaseDate: '2024-12-19T20:40:41.663Z'
    }
  },
  ports: []
}
[bluetooth] Device Detected | 8E871878-D780-89EE-AF6B-8B9B27AB4DD5 G2_d85c
API PROCESS EXITED! 6
```

jwarchol, 2025-01-10:
Basically the only way I can get MeshSense to actually run is to disable bluetooth. That's useful for pulling down updates, but not so useful for connecting to most nodes.

Soltares, 2025-01-10:
Thanks for the update.  I did some digging and it looks like this issue is [possibly fixed](https://github.com/simpleble/simpleble/issues/277#issuecomment-2541756994) in version 0.8.1 of `simpleble`.  It will require a rebuild of `webbluetooth` with the updated version and to make sure that we are in compliance with [new license ](https://github.com/simpleble/simpleble/blob/main/LICENSE.md) of simpleble.

jwarchol, 2025-01-10:
Since MeshSense is open source under GPLv3 and `simpleble` is too, hopefully it's a non-issue. If needed they do seem to have an option for a free commercial license:

> Likewise, if you are using SimpleBLE in an open-source project and would like to request a free commercial license or if you have any other questions, please reach out at contact at simpleble dot org.

Appreciate the update. I'm still able to enjoy MeshSense by running it on a PC and accessing it via a web browser on my mac, but I'm really looking forward to being able to run it on MacOS too here.

tavdog, 2025-04-01:
Upgraded M1 Macbook to Seqoia.  Meshsense ran for a few days, then I upgraded and it ran for one more day. I ran it for an hour or so offline and then it suddently would not start.  Tried deleting application data and reinstalling. Tried with BT off and WiFi off.  Following is the error when starting from terminal : ./MeshSense --trace-warnings                 
DIRNAME /Applications/MeshSense.app/Contents/Resources/app.asar/out/main
API_PATH /Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [
  '/Applications/MeshSense.app/Contents/MacOS/MeshSense',
  '--trace-warnings'
]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api',
  staticDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/static',
  dataDirectory: '/Users/tavis/Library/Application Support/meshsense'
}
undefined:1

SyntaxError: /Users/tavis/Library/Application Support/meshsense/state/address: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Object.readFileSync (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:3556:18)
    at Object.getSync (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:9578:42)
    at Object.get (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:9867:29)
    at new State (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:10329:36)
    at Object.<anonymous> (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:84365:15)
    at Module._compile (node:internal/modules/cjs/loader:1484:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1564:10)
    at Module.load (node:internal/modules/cjs/loader:1295:32)
    at Module._load (node:internal/modules/cjs/loader:1111:12)

Node.js v20.18.0
API PROCESS EXITED! 1

Soltares, 2025-04-01:
Looks like an eventual crash is still possible while searching for bluetooth devices:

```
[bluetooth] scanning...
[bluetooth] Device Detected | 01D72BD5-4865-E19A-9140-C61D698DF67E Unknown or Unsupported Device (01D72BD5-4865-E19A-9140-C61D698DF67E)
[bluetooth] scanning...
API PROCESS EXITED! 11
```

tavdog, 2025-04-02:
> Thanks for the report [@tavdog](https://github.com/tavdog). Based on the log, it looks like somehow the saved value for `address` became malformed. An easy fix would be to delete that file:
> 
> ```
> /Users/tavis/Library/Application Support/meshsense/state/address
> ```
> 
> The original problem with this issue seemed to center around bluetooth controller related crashes. I'm optimistic that the 1.1.0-beta series has resolved this issue as I have run `v1.1.0-beta.3` for multiple days now using bluetooth on my M2 Sonoma 14.7.1

deleting the application support dir does nothing 
`tavis@M1Pro /A/M/C/MacOS> rm -rf /Users/tavis/Library/Application Support/meshsense                                                                                                                1
tavis@M1Pro /A/M/C/MacOS> ./MeshSense                                              
DIRNAME /Applications/MeshSense.app/Contents/Resources/app.asar/out/main
API_PATH /Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [ '/Applications/MeshSense.app/Contents/MacOS/MeshSense' ]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api',
  staticDirectory: '/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/static',
  dataDirectory: '/Users/tavis/Library/Application Support/meshsense'
}
undefined:1

SyntaxError: /Users/tavis/Library/Application Support/meshsense/state/address: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Object.readFileSync (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:3556:18)
    at Object.getSync (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:9578:42)
    at Object.get (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:9867:29)
    at new State (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:10329:36)
    at Object.<anonymous> (/Applications/MeshSense.app/Contents/Resources/app.asar.unpacked/resources/api/index.cjs:84365:15)
    at Module._compile (node:internal/modules/cjs/loader:1484:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1564:10)
    at Module.load (node:internal/modules/cjs/loader:1295:32)
    at Module._load (node:internal/modules/cjs/loader:1111:12)
`

Soltares, 2025-04-18:
I am continuing to receive reports of MacOS crashing related to bluetooth even while using 1.1.0-beta.5.  This beta is using version 0.9.1 `0e510bca015455cd45860a368068d631369dc5d2` of SimpleBLE which reportedly was supposed to fix the full-length UUID issue.  I'm a bit at a loss on how to proceed.

> Thanks for the update. I did some digging and it looks like this issue is [possibly fixed](https://github.com/simpleble/simpleble/issues/277#issuecomment-2541756994) in version 0.8.1 of `simpleble`.

tavdog, 2025-04-19:
My system crashes everytime now. I can't even run the program.

-tavis

Jutral, 2025-08-06:
So, I run MeshSense on a few Mac devices. My M4 MacBook Pro has decided that it will no longer open MeshSense. I've completely deleted/reinstalled the app, but if I have Bluetooth enabled--the app will immediately crash on open.

Happens on 1.0.18 and the latest beta (beta.7).

MacOS 15.6.