Affirmatech/MeshSense #18: Trace Route Rate Limiting in New Firmare

garthvh, 2024-09-27:
Very cool app, glad to see it open sourced!

Rate limiting was added to the latest firmware, take a look at the ClientNotification protobuf it will alert you when you have hit the rate limit (30 seconds)

Soltares, 2024-09-27:
Thanks for letting us know!  We'll take a look!

tcrosb, 2024-10-02:
@garthvh Do you know what build the trace route was limited in?

garthvh, 2024-10-02:
This release https://github.com/meshtastic/firmware/releases/tag/v2.5.1.c13b44b

Soltares, 2024-10-24:
I looked at upgrading `@meshtastic/js` from 2.3.7-2 to 2.5.5-0 as the latest commit migrates to a new protobufs package:

```
"@meshtastic/protobufs": "npm:@jsr/meshtastic__protobufs@^2.5.7",
```

I was hoping this would add a definition for ClientNotification, but it did not appear to do so as shown in the log:

```
18:06:07:434    TRACE   [iMeshDevice:HttpConnection]    HandleFromRadio 🚧 Received Queue Status: [object Object]
18:06:07:503    WARN    [iMeshDevice:HttpConnection]    HandleFromRadio ⚠️ Unhandled payload variant: clientNotification
```

The format of the incoming data is different as well, as `type` is now `$typeName` and creates "undefined" entries in the packet log:

```
{
  "$typeName": "meshtastic.MeshPacket",
  "from": 4087226095,
  "to": 4294967295,
  ...
```

For now I will keep the working  `@meshtastic/js` 2.3.7-2 version and assume a global trace route limit of 30 seconds.

closed 2024-11-25 by Soltares