Affirmatech/MeshSense #22: Support for arm64 Raspberry Pi

Soltares, 2024-09-27:
A few requests have come in requesting to run MeshSense on a Raspberry Pi.  This should be possible, in theory, although the `webbluetooth` library does not have a prebuild currently for arm64.  See https://github.com/thegecko/webbluetooth/issues/190

Will need to look to see how a build would be accomplished manually.

RyanWor, 2024-09-27:
Subscribed. Thank you Chris! 

I guess I should have just said arm, not sure if changes things or matters. I'd be perfectly fine running 32bit OS if possible since it would be the Pi only role. Either though way thanks for this awesome application! I am loving it!

azwirko, 2024-10-29:
I'll admit this is surely NOT elegant.  I'm not using BT to connect to my node, and only access via WiFi IP. To that extent I've hacked a few of the "api" related .ts files to bypass or ignore some of the BT functions.  I simply ran the code and commented out sections until I could get a response on http://localhost:5920/  I've not extensively tested MeshSense doing this, but the attached screenshot shows its at least working to an extent on my RPi 4. Maybe a 'diff' export could be made or possible a #ifdef block for "noBT" to accomplish the same.

Aside from having to 'apt-get install' cmake and ninja-build, here are the changes made to some of the MeshSense files prior to starting the api and ui pieces with npm.

in api/packages.json 
   line 55 delete:
```
   "webbluetooth": "^3.2.1",

```

in api/src/meshtastic.ts
   line 26 comment out:
```
   // import { beginScanning, bluetoothDevices, stopScanning } from './lib/bluetooth'
```
   lines 54-58 rewrite as:
```
   if (value == 'disconnected' || value == 'searching' || value == 'reconnecting') {
      // beginScanning()
   } else {
      // stopScanning()
   }
```

in api/src/lib/bluetooth.ts 
   line 1 comment out:
```
   // import { Bluetooth } from 'webbluetooth'
```

[image]

Happy hacking!

Mesh node K1RA 
LongFast Slot 9 in Northern Virginia west of Washington D.C.

Soltares, 2024-10-29:
That's awesome!  I would anticipate MeshSense to largely work without issue.  It was originally built with IP-only and bluetooth was added later.  Only thing that stands out to me are the lack of traceroute lines in the screenshot, but the connected node is not reporting a position so that could be part of the reason.

This brings up an interesting question of whether or not to provide an official ARM build with bluetooth functionality disabled.

Thanks for sharing! 👏

azwirko, 2024-10-30:
Soltares, good catch.  My dev node 1RA2 does not have built-in GPS, nor Lat/Lon set.  After setting up a fixed position now the map shows as attached:

[image]

Hope you can get it running as well.

Soltares, 2025-02-03:
Just wanted to provide a quick update as I've been receiving frequent requests for this feature.

Steps taken are as follows:

- Created `webbluetooth` fork to use latest `SimpleBLE` version.  (to possibly also fix MacOS issues)
- Added `webbluetooth` fork as a submodule for MeshSense so that it can be built on local hardware.
- Updated `@meshtastic/js` to version 2.5.9-3 which actually removes the `serialport` dependency and simplifies API rollup build process.
- Tested on a Raspberry Pi 4 and Pi OS and **MeshSense starts**!

The problem, however, is that there are changes that come along with the latest `@meshtastic/js` version that need to be accounted for.  Particularly the event types (like POSITION_APP) are not quite named the same way.  I am working through these as my schedule allows and hope to have good news soon.

Cisien, 2025-02-23:
I'm looking forward to running this on the pi I just set up with a lora hat and meshtasticd. Please let me know if I can be of any assistance with testing/validation.

Soltares, 2025-02-23:
Excellent!  Last week I was able to build on the Pi successfully.  Let me run a fresh build with the latest commit and I will post the link here for testing.  Thanks!

Soltares, 2025-02-24:
Beta AppImages of the following ARM builds are available if anyone wanted to give it a try:

https://affirmatech.com/download/meshsense/meshsense-beta-arm64.AppImage (64-bit)
https://affirmatech.com/download/meshsense/meshsense-beta-armv7l.AppImage (32-bit)

Cisien, 2025-02-25:
```
(meshtastic) cisien@hamclock:~ $ ACCESS_KEY=meshsense ./meshsense-beta-arm64.AppImage --headless
-bash: ./meshsense-beta-arm64.AppImage: cannot execute: required file not found
```

Raspbian 12.9 64-bit on a pi 4, fully updated as of Sunday.

Cisien, 2025-02-25:
I got it to work by creating a ~/.local/share/meshsense/state/enableTLS file with the contents of `true`!

[image]

Cisien, 2025-02-25:
I have this running with systemd as well, feel free to ship this .service file somewhere:

```
[Unit]
Description=MeshSense Headless Daemon
After=multi-user.target

[Service]
Type=simple
Environment="ACCESS_KEY=meshsense"
ExecStart=/path/to/meshsense.AppImage --headless
Restart=always
User=meshsense

[Install]
WantedBy=default.target
```

OSRdesign, 2025-02-25:
[image]

I had to manually pair my node in Bluetooth from terminal as from the app or piOS it was not triggering the passcode entry. But then works like a charm so far. 
Arm64 version on CM4 on this custom piOS for the uConsole.

Soltares, 2025-02-25:
Thank you @Cisien and @OSRdesign for the good reports!

Adding another test result fom a user running on Bullseye:

```
Error: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found
```

This is similar to #59 where simpleble is pulling newer dependencies when building on Bookworm.  I will likely need to build on Bullseye for broader support.

Famicoman, 2025-02-25:
@Soltares FWIW I get that same error on my x64 Bullseye machine.

BergischNRW, 2025-02-26:
The beta build for arm64 requires quite a bit of packages even when you want to run it headless. Is that normal?
`/tmp/.mount_meshsehLui5w/meshsense: error while loading shared libraries: libatk-1.0.so.0: cannot open shared object file: No such file or directory`

`/tmp/.mount_meshseJRtBkI/meshsense: error while loading shared libraries: libatk-bridge-2.0.so.0: cannot open shared object file: No such file or directory`

`/tmp/.mount_meshsecUYWqM/meshsense: error while loading shared libraries: libgdk_pixbuf-2.0.so.0: cannot open shared object file: No such file or directory`

`/tmp/.mount_meshseXblLkU/meshsense: error while loading shared libraries: libgtk-3.so.0: cannot open shared object file: No such file or directory`

BergischNRW, 2025-02-26:
After installing `libgtkd-3-0` and `libgbm1` I´m getting the following error: 
`Trace/breakpoint trap`

I can confirm that the `~/.config/meshsense` folder was created but it´s empty.

mikecarper, 2025-02-27:
Got it working with a pi zero w 2
```bash
wget https://affirmatech.com/download/meshsense/meshsense-beta-arm64.AppImage
dbus-run-session xvfb-run ./meshsense-beta-arm64.AppImage --headless --disable-gpu
```

BergischNRW, 2025-02-27:
I also got it working now with the `dbus-run-session xvfb-run ./meshsense-beta-arm64.AppImage --headless --disable-gpu` command. Thanks @mikecarper!

joeldhenry, 2025-02-28:
wouldnt run for me without doing what @mikecarper said. plus still needs lots of dependancies installed.

RyanWor, 2025-03-02:
Does anyone have details on what packages were necessary? I've tried this on Bookworm and Bullseye on a 2 W and still get the following error:
`Error: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_meshsek5z4WH/resources/app.asar.unpacked/resources/api/simpleble.node)
    at process.func [as dlopen] (node:electron/js2c/node_init:2:2559)
    at Module._extensions..node (node:internal/modules/cjs/loader:1602:18)
    at Object.func [as .node] (node:electron/js2c/node_init:2:2559)
    at Module.load (node:internal/modules/cjs/loader:1295:32)
    at Module._load (node:internal/modules/cjs/loader:1111:12)
    at c._load (node:electron/js2c/node_init:2:16955)
    at Module.require (node:internal/modules/cjs/loader:1318:19)
    at require (node:internal/modules/helpers:179:18)
    at Object.<anonymous> (/tmp/.mount_meshsek5z4WH/resources/app.asar.unpacked/resources/api/index.cjs:76756:30)
    at Module._compile (node:internal/modules/cjs/loader:1484:14) {
  code: 'ERR_DLOPEN_FAILED'
}

Node.js v20.18.0
API PROCESS EXITED! 1
`

Soltares, 2025-03-02:
> This is similar to [#59](https://github.com/Affirmatech/MeshSense/issues/59) where simpleble is pulling newer dependencies when building on Bookworm. I will likely need to build on Bullseye for broader support.

@RyanWor I built this on Bullseye and need to rebuild on Bookworm so that it doesn't pull a newer dependency than what's available for Bookworm.  I am trying to build a VM to accomplish this, but it's taking longer than expected.

Cisien, 2025-03-14:
@Soltares I'm not sure if this is related to this fork, but sending messages via meshsense with firmware 2.6 (preview) sets the hops to 0/0 (total/remaining)

jbouse, 2025-03-28:
I just wanted to share my [Gist](https://gist.github.com/jbouse/1fc21bb3d6be4409054bf079d6cb57bd) for my headless setup on my Raspberry Pi, which I've been running since beta.1 and just tested with auto-restart of beta.3 download through the UI. I believe the SystemD unit file will handle running all three Linux environments (x86_64, beta-arm64, and beta-armv7l), but I've only tested beta-arm64 on both my Raspberry Pi 4 and Pi 5.

corrt, 2025-08-07:
Not sure if this is the right place to ask. Is there any chance that the 1.0.0-beta7 arm64 AppImage is corrupted?
This is what I'm seeing (running Buster):

 - $ wget https://affirmatech.com/download/meshsense/meshsense-beta-arm64.AppImage
 - $ sudo chmod a+x ./meshsense-beta-arm64.AppImage
 - $ sudo ./meshsense-beta-arm64.AppImage 
    - Syntax error: "(" unexpected
 - $ ./meshsense-beta-arm64.AppImage 
    - bash: ./meshsense-beta-arm64.AppImage: cannot execute binary file: Exec format error

Outrun207, 2026-02-17:
## Bluetooth Scanning Crashes on Raspberry Pi 3 Model B (ARM64)

I successfully built MeshSense from source on a Raspberry Pi 3 Model B (1GB RAM,
Debian 11 Bullseye, aarch64) following the development setup instructions. The 
build process completed after upgrading to CMake 3.27, and both the UI and API 
services start correctly.

Environment:
- Raspberry Pi 3 Model B Rev 1.2
- Debian GNU/Linux 11 (bullseye)
- Node.js v18.20.8
- CMake 3.27.7
- Bluetooth 5.55 (working, verified with bluetoothctl)

Issue:
The API starts and attempts Bluetooth scanning, but crashes after 10 seconds 
with:

[bluetooth] Begin Scanning
[bluetooth] Adapter available: true
[bluetooth] scanning...
[system] Uncaught Exception Error: scan stop failed
    at SimplebleAdapter.stopScan (/home/josh/MeshSense/api/webbluetooth/src/adapters/simpleble-adapter.ts:179:23)
    at BluetoothImpl.cancelRequest (/home/josh/MeshSense/api/webbluetooth/src/bluetooth.ts:396:21)
    at Timeout.<anonymous> (/home/josh/MeshSense/api/webbluetooth/src/bluetooth.ts:307:22)

This happens consistently on every scan attempt. The Pi's Bluetooth hardware is 
functional (verified with bluetoothctl and hcitool), and I've confirmed:
- User is in the bluetooth group
- Node has cap_net_raw,cap_net_admin capabilities set
- Multiple Meshtastic nodes with Bluetooth enabled are nearby (but none are 
detected)

Observations:
- The webbluetooth library builds successfully with the SimpleBLE dependency
- The crash occurs when trying to stop the scan, not when starting it
- No Bluetooth devices are detected before the crash
- The UI loads correctly but shows no Bluetooth connection option

It appears the SimpleBLE adapter has runtime issues on ARM64 Raspberry Pi 
hardware, even though compilation succeeds. This might be related to the 
underlying SimpleBLE library's compatibility with the Pi's Bluetooth chipset.

Has anyone successfully run MeshSense with Bluetooth on a Raspberry Pi 3 or 4? 
Any workarounds or alternative Bluetooth libraries that might work better on 
ARM64?