Affirmatech/MeshSense #23: Feature: UI/UX Pass

nbgibson, 2024-09-30:
The app in its current state is in a pretty good place I feel. The UI is largely self explanatory as mentioned in the website's FAQs, but there are inscrutable elements that could be shored up. I think adding "on hover" elements to more of the app's assorted symbols and so on saying what they are would be a massive quality of life improvement for the average user.

We need to keep in mind that Meshtastic is getting taken up by folks typically not within the HAM-sphere so stuff that may be largely self evident to people versed in the lifestyle may as well be hieroglyphics to someone just coming in. 

EG: The Cs or Rs you will occasionally get beside node names. I presume the former to be a indicator of a Constant set position, but I have no real way of know. I do appreciate that we do have stuff for "fly to", "trace route", "request position", and so on; we just need that for the rest of the icons.

Soltares, 2024-09-30:
Awesome feedback, thank you!  Come to think of it I don't believe we mention the C and R icons in the FAQ either.  They currently represent "Client" vs "Router" nodes and both for "CLIENT_ROUTER" (which I believe is going away in recent firmware versions)

Regardless we could definitely go through and make sure everything has an appropriate tooltip or help text to clear some of this up!

Soltares, 2024-10-22:
Closing for now as #24 is included in latest release