Affirmatech/MeshSense #25: Not able to find Environmental Telemetry logs

jasgroupit, 2024-10-11:
I have several nodes setup with one main node to view the data of my other nodes which i use in a farm. but i am not able to see telemetry app logs regarding environmental data

Kealper, 2024-10-17:
I'm just guessing but they may be talking about the telemetry coming from the environmental and power metrics in the Telemetry module. Stuff like temperature, humidity, pressure, current draw, etc., from I2C sensors that can be added.

jasgroupit, 2024-10-21:
Yes exactly that, temperature humidity pressure etc. Is there a way to display this ?

lupusworax, 2024-11-18:
Same Request and issue.
As of now a node that functions perfectly with a BME280  shows the following:

{
  "num": 3306578313,
  "user": {
    "id": "!c5166189",
    "longName": "Lupus Worax 6189",
    "shortName": "🐺",
    "macaddr": "5NrFFmGJ",
    "hwModel": "HELTEC_MESH_NODE_T114",
    "isLicensed": false,
    "role": "CLIENT",
    "publicKey": "R3Lx2m/TVSMQDLASID=8363dDvaQdmJrg8+Ho="
  },
  "position": {
    "latitudeI": 477885538,
    "longitudeI": 137496299,
    "altitude": 210,
    "time": 1731941595,
    "locationSource": "LOC_UNSET",
    "altitudeSource": "ALT_UNSET",
    "timestamp": 0,
    "timestampMillisAdjust": 0,
    "PDOP": 0,
    "HDOP": 0,
    "VDOP": 0,
    "gpsAccuracy": 0,
    "groundSpeed": 0,
    "groundTrack": 0,
    "fixQuality": 0,
    "fixType": 0,
    "satsInView": 0,
    "sensorId": 0,
    "nextUpdate": 0,
    "seqNumber": 0,
    "precisionBits": 32
  },
  "snr": 6.5,
  "lastHeard": 1731941591.13,
  "deviceMetrics": {
    "batteryLevel": 78,
    "voltage": 3.9730000495910645,
    "channelUtilization": 4.791666507720947,
    "airUtilTx": 0.2473333328962326,
    "uptimeSeconds": 98516
  },
  "channel": 0,
  "viaMqtt": false,
  "hopsAway": 0,
  "isFavorite": false,
  "rssi": -53,
  "trace": null
}

Soltares, 2024-11-18:
Environmental data appears to come from the `environmentMetrics` Telemetry Message type.  Will attempt to capture this and attach to the tracked Node information.

Soltares, 2024-11-20:
Successfully captured some data with the latest dev build!

```json
  "environmentMetrics": {
    "temperature": 15.008466720581055,
    "relativeHumidity": 53.141929626464844,
    "barometricPressure": 1004.330810546875,
    "gasResistance": 624.9585571289062,
    "iaq": 133
  }
  ```

lupusworax, 2024-11-20:
> Successfully captured some data with the latest dev build!
> 
> ```json
>   "environmentMetrics": {
>     "temperature": 15.008466720581055,
>     "relativeHumidity": 53.141929626464844,
>     "barometricPressure": 1004.330810546875,
>     "gasResistance": 624.9585571289062,
>     "iaq": 133
>   }
> ```

Thats freaking awesome! 💯   cant wait to see it pop up with the next ( hopefully)  update

Soltares, 2024-11-22:
Initial draft at displaying the data
[image]
Temp / Pressure / Gas Resistance / Humidity / Air Quality

lupusworax, 2024-11-22:
> Initial draft at displaying the data [...]

You nailed it!

Soltares, 2024-11-22:
This has been published in `v1.0.16-beta.2`!  Hopefully everything works well!

lupusworax, 2024-11-22:
updated but so far no telemetry data from my solar node, could it take a while until it updates telemetry?

Soltares, 2024-11-22:
Yes it's possible.  There was only one node in my local mesh that published data and it was on the edge of my network.  It is likely based on the environmentUpdateInterval setting of your solar node.

Device configuration of the MeshSense connected node can now be viewed in the Settings modal:

[image]

lupusworax, 2024-11-22:
I see, yeah the connected node sadly does not have an environmental sensor. Well I give it a bit and will report back, as of now it looks like this: 
[image]

lupusworax, 2024-11-22:
Fuck yeah it works! 
[image]
amazing work! Thanks for taking the time to work  and adding this!

Soltares, 2024-11-22:
Fantastic!!  Thanks for the report!  Looks like I need to potentially hide or clean up fields if they're missing.  Appreciate it!

Soltares, 2024-11-25:
Closing as feature is released in `v1.0.16`

closed 2024-11-25 by Soltares