Affirmatech/MeshSense #27: [Feature] Remember last map position and zoom view - dont reset/default back to North America

Talie5in, 2024-10-22:
Remember last map position and zoom view - dont reset/default back to North America

This is annoying to come back to a machine and its looking at North America, or when re-opening the app, default backs to North America.

Store last view and return on app start etc.

Soltares, 2024-10-23:
That is an excellent suggestion, thanks!  By default it will center on the position of the connected node, but if that is not available it zooms out to North America.  Reloading last position instead makes a lot of sense.

Soupbones, 2024-11-07:
I came here to say this exact thing as I just started usnig this awesome app to monitor my local mesh. Its dam annoying to have to find your node location every time you restart the app. If your node had a location then the app shoudl automatically start there on the map. I would also like to see messaging done better, maybe a window or a tab for messages from nodes you are communication with.

Soltares, 2024-11-07:
Thanks @Soupbones for the feedback!  The latest beta `v1.0.16-beta.1` should have the map position save feature if you wanted to check it out.  Simply enable Beta Updates inside of settings.  Thanks!

Soupbones, 2024-11-07:
> Thanks @Soupbones for the feedback! The latest beta `v1.0.16-beta.1` should have the map position save feature if you wanted to check it out. Simply enable Beta Updates inside of settings. Thanks!

Great to get such a quick reply. Not often you see open source developers working to fix things so quick. I commend you!
 That said, the update feature worked great. I very quickly had the new beta installed. Alas, now I have no map. The map doesnt seem to be loading at all now.  I may have to uninstall to roll it back. Using Windows 11.

Soltares, 2024-11-25:
> The map doesnt seem to be loading at all now.

I have seen this happen very rarely, but usually a reconnect or application restart will resolve the issue.  The map does require internet access to download the tiles.

Soltares, 2024-11-25:
Closing at the requested feature has been released in `v1.0.16`

closed 2024-11-25 by Soltares

Soupbones, 2024-11-25:
I did that. Restart several times, reinstall. Still did it. Gave up, went
back to previous version. My PC has working internet.