Affirmatech/MeshSense #30: Error installing under Linux

mdede439, 2024-11-08:
I am attempting to install MeshSense on a Debian VM on Proxmox and I am running into an error when I run  ./update.mjs.  It returns and error 

import { styleText } from 'util';
SyntaxError: The requested module 'util' does not provide an export named 'styleText'

Have you run into this?  Where is module 'util'?

mdede439, 2024-11-08:
I figured out my issue.  I had no idea what a .mjs file was.  after figuring out that, it turns out I installed too low a version of Node.js.  Got it installed now.

closed 2024-11-08 by mdede439

mikecarper, 2025-02-13:
```
node -v
npm -v
curl -fsSL https://deb.nodesource.com/setup_20.x | sudo -E bash -
sudo apt-get install -y nodejs
node -v
npm -v
```