Affirmatech/MeshSense #34: strange crash - ReadFromRadio ❌ Cannot read properties of undefined (reading 'buffer')

lagunacomputer, 2024-11-28:
Any idea why this happens about every 30-60 mins?   Drops BT connection, haven't checked yet if the node reboots (its on the roof).  meshtastic 2.5.14 on a tbeam clone DIY node, thanks for lookin'

04:22:53:229 INFO [iMeshDevice:HttpConnection] HandleFromRadio 📱 Received Node Info packet for node: 3806455808
**04:22:53:470 WARN [iMeshDevice:HttpConnection] HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
04:22:53:830 WARN [iMeshDevice:HttpConnection] HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
04:22:54:068 WARN [iMeshDevice:HttpConnection] HandleFromRadio ⚠️ Unhandled payload variant: fileInfo
04:22:54:370 WARN [iMeshDevice:HttpConnection] HandleFromRadio ⚠️ Unhandled payload variant: fileInfo**
04:22:54:490 INFO [iMeshDevice:HttpConnection] HandleFromRadio ⚙️ Valid config id received from device: 943727779
[meshtastic] Device Status 7
[State] connectionStatus set connected
[meshtastic] Device Status 5
04:22:55:031 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:22:55:212 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:22:55:631 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received POSITION_APP packet
Packet 200841686 of type wantConfigId timed out
04:23:48:370 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:23:57:372 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:24:47:772 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:24:47:889 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:25:48:252 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:26:13:452 DEBUG [iMeshDevice:HttpConnection] HandleMeshPacket 🔐 Device received encrypted data packet, ignoring.
04:26:43:332 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received POSITION_APP packet
04:26:47:890 TRACE [iMeshDevice:HttpConnection] HandleFromRadio 🚧 Received Queue Status: [object Object]
04:26:48:312 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
04:27:03:490 TRACE [iMeshDevice:HttpConnection] HandleFromRadio 🚧 Received Queue Status: [object Object]
04:27:04:089 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received POSITION_APP packet
**04:27:58:180 ERROR [iMeshDevice:HttpConnection] ReadFromRadio ❌ Cannot read properties of undefined (reading 'buffer')
04:27:58:589 ERROR [iMeshDevice:HttpConnection] ReadFromRadio ❌ Cannot read properties of undefined (reading 'buffer')
04:27:58:590 INFO [iMeshDevice:HttpConnection] Connect Device disconnected
[meshtastic] Device Status 2
Connection Intended true
[bluetooth] Begin Scanning
[State] connectionStatus set reconnecting
[meshtastic] Calling connect cc:50:e3:a9:34:6a
[bluetooth] Begin Scanning
[State] connectionStatus set disconnected
Disconnecting from device
[bluetooth] Adapter available: true
[bluetooth] scanning...
[bluetooth] Adapter available: true
[bluetooth] scanning...
[bluetooth], Error encountered during scan Error: requestDevice error: request in progress
at BluetoothImpl.requestDevice (C:\Users\gman\AppData\Local\Programs\meshsense\resources\app.asar.unpacked\resources\api\index.cjs:87082:20)
at scanForDevice (C:\Users\gman\AppData\Local\Programs\meshsense\resources\app.asar.unpacked\resources\api\index.cjs:107664:38)
at beginScanning (C:\Users\gman\AppData\Local\Programs\meshsense\resources\app.asar.unpacked\resources\api\index.cjs:107708:9)
at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
04:27:58:610 DEBUG [iMeshDevice:HttpConnection] Constructor 🔷 BleConnection instantiated
Already Scanning
[State] connectionStatus set searching
Already Scanning**
[bluetooth] Device Detected | 55:82:bd:84:81:60 Unknown or Unsupported Device (55:82:bd:84:81:60)
[bluetooth] scanning...
[bluetooth] Device Detected | 78:7a:8c:1b:ac:43 Unknown or Unsupported Device (78:7a:8c:1b:ac:43)
[bluetooth] scanning...
[bluetooth] Device Detected | 78:7a:8c:1b:ac:43 Unknown or Unsupported Device (78:7a:8c:1b:ac:43)
[bluetooth] scanning...
[bluetooth] Device Detected | 55:82:bd:84:81:60 Unknown or Unsupported Device (55:82:bd:84:81:60)
[bluetooth] scanning...
[bluetooth] Device Detected | 78:7a:8c:1b:ac:43 Unknown or Unsupported Device (78:7a:8c:1b:ac:43)
[bluetooth] scanning...
[bluetooth] Device Detected | 78:7a:8c:1b:ac:43 Unknown or Unsupported Device (78:7a:8c:1b:ac:43)
[bluetooth] scanning...
[bluetooth] Device Detected | 78:7a:8c:1b:ac:43 Unknown or Unsupported Device (78:7a:8c:1b:ac:43)
[bluetooth] scanning...
[bluetooth] Device Detected | cc:50:e3:a9:34:6a 4Rch_3468
[bluetooth] Stop Scanning
[State] connectionStatus set connecting
[meshtastic] Connecting to Node cc:50:e3:a9:34:6a via Bluetooth
[meshtastic] Device Status 3

defsix, 2024-12-04:
I've the same issue with my Heltec V2,  no problem with TBeam and TEcho

Eric-Sparks, 2025-02-27:
I believe I'm seeing the same problem when attaching to my RAK 4631.  It will usually stabilize and stay connected for a while but other times (like right now) it just cycles.

christopher-oswald, 2025-03-28:
Same issue on my end. Windows 11, Dell Latitude 7390. Device is a Rak4631. 

Confirmed that my device does work with BT w/ the official browser client.

cco, 2025-03-28:
Same issue on Windows 11, Dell Latitude 7390. 

Confirmed Bluetooth works on the official web browser client from Meshtastic.