Affirmatech/MeshSense #36: Unable to connect to T-Deck Plus via IP

IamTBUCK, 2024-12-06:
Attempting to connect to the T-Deck Plus via IP (both with TLS On and Off) and get the following results:

[meshtastic] Device Status 3
19:04:22:910 DEBUG [iMeshDevice:HttpConnection] Ping Attempting device ping.
19:04:32:921 ERROR [iMeshDevice:HttpConnection] Ping ❌ fetch failed
[meshtastic] Device Status 4

I am able to connect to the device from the Meshtastic mobile app via IP.

IamTBUCK, 2024-12-07:
I have an update.

It appears that the new MeshUI (fancy gui firmware) interferes with this. I flashed the stock firmware and can connect via IP. Issue resolved, but will likely be an issue for others that choose to use the fancy UI moving forward

Soltares, 2025-01-03:
Thanks for the report and follow-up!  We will be on the lookout for MeshUI interaction troubles in the future :+1:

closed 2025-01-03 by Soltares