Affirmatech/MeshSense #37: Lines on map to nodes not working on RAK

Soupbones, 2024-12-08:
I'm loving Meshsense for monitoring my local n etwork as I have added a couple of remote nodes and its great to keep an eye on them. I noticed with my RAK 4631/19007 that there are never lines showing up on the map when it does a trace route. Ever.
 So I tested this on my Heltec V3 and staright away it shows line to nodes its connecting too. I've done db clearing and it still won't show the lines on the map.
[image]
[image]

 Anyway to fix this?
The Heltec is using Firmware 2.5.9 and the RAK is using 2.5.13

jwarchol, 2024-12-12:
I've also seen the lines not showing, sometimes, on several type of nodes (such as a T-Beam Supreme and a Station G2)

Soltares, 2024-12-16:
Interesting -- our [demo node](https://affirmatech.com/meshsense/FJP1/) happens to be RAK4631.  If you click :mag: on your connected node, does it report a position?

I might also verify using the Meshtastic app that a region has been selected.

KB1RMA, 2025-05-05:
I just started running into this same issue but instead on a t-beam. Took a quick look at the code and it looks like this issue (at least for me) happens when Meshsense doesn't know what the home coordinates of the node its connected to are.

Here's the problematic code: https://github.com/Affirmatech/MeshSense/blob/master/ui/src/Map.svelte#L36

When `let myNodeCoords = getCoordinates($myNodeNum)` returns `[undefined, undefined]` the resulting array passed to `plotLines()` which in turn passes it to `new LineString(points)` from `ol` looks like: 
```ts
[
	[undefined, undefined],
	[-70.8706304, 42.8212224],
	[-70.8902912, 42.8146688],
]
```

Loosening this check to make sure we're not passing falsy values to the lines on the map fixes it up:

```ts
          return list.filter((coords) => (coords[0] && coords[1]))
```

Going to test this a bit more and read up on the codebase a bit then I'll submit a PR

Soltares, 2025-05-07:
Good catch!  I'm surprised traceroutes are still as informative as they are despite the node that initiated them being unmapped.

Another possible code implementation:

```ts
        .map((n) => {
          let list: any[] = [...(n.trace?.route?.map((traceNode) => getCoordinates(traceNode)) || []), getCoordinates(n)]
          if (myNodeCoords[0] && myNodeCoords[1]) list.unshift(myNodeCoords)
          return list.filter((coords) => !(coords[0] == 0 && coords[1] == 0))
        })
```

Looking at the `list.filter` has me wondering if it is doing more harm than good.  Right now, if there is a node within a trace that does not have a position, MeshSense would in theory still draw a line in between it's neighbors even though they are not directly connected.  (Granted this scenario is lessened by the use of approximate locations)

For the global map, if there is any node in a trace that does not have a position (including the source node), I skip drawing the entire trace route to avoid this situation.  I wonder if we consider this sort of filter for MeshSense prior to `.map`.

I could see a possible compromise being "trimming" the start and end of the trace route for nodes with no positions so that we still get some valid connections drawn.

KB1RMA, 2025-05-07:
So far in my testing my example code has been doing fine. I, generally speaking, always prefer to avoid array mutation with things like `unshift` especially within a `.map` itself but I wanted to take a broader look at the flow before I submit a patch. The types for `NodeInfo` appear to be pretty far off from reality so I was going to stick my nose into it further to prevent this type of bug elsewhere. Still working on finding time to do that! 

I'm new to this codebase but love the software and am a TS fella professionally so looking forward to trying to get myself acquainted!

Soltares, 2025-05-07:
Awesome!!  As a humble Typescript enthusiast that's learning as I go, I am more than happy to defer to your experience!  (I hope my codebase isn't too egregious 😆 )

madasus, 2025-09-18:
Was this ever fixed in code? I'm also seeing this and i'm not sure how to fix.