Affirmatech/MeshSense #38: Traceroute node path text shown incorrectly when receiving a traceroute request.

jwarchol, 2024-12-16:
[image]

[image]

I sent a traceroute from CARD (1139140003) to JW+ (1129837892). MeshSense 1.0.18-beta shows the path with the ends transposed. 

Incorrect current output: `JW+ -> FILT -> GFR2 -> SOL -> CARD`
Correct output: `CARD -> FILT -> GFR2 -> SOL -> JW+`

Here is the raw data as shown when clicking the magnifying glass in MeshSense: https://gist.github.com/jwarchol/3af4f71d116ab8241795c152ae0d8ccc

jwarchol, 2024-12-16:
To reiterate, this is when the trace is from the other node (CARD) to the node connected to MeshSense (JW+ here)

Soltares, 2024-12-16:
This is interesting thanks for reporting this.  I hadn't realized trace routes issued by other nodes would appear as packets in the connected node.  This somewhat shakes up the way I store trace routes as I am currently attaching the route to the node data for the destination node.  In this case the destination node is the connected node, but different nodes could send it.  I suppose a quick solution would be pick the node that's not the current node to attach it to (although the route could be in reverse).  Thinking out loud :smile: