Affirmatech/MeshSense #40: plotting MQTT links on map: not useful and degrading performance.

recrof, 2024-12-17:
MQTT links especially on global map are doing more harm than good - there should be a way to disable or disallow them.

[image]

If there is no way to detect MQTT links, MeshSense should ignore links that are longer than ~300-600 km.

Soltares, 2024-12-17:
Hi and thanks for report.  The Global MeshSense map is currently under active development and we are trying to find the best solution to this problem.  We believe we have filtered out MQTT routes already in f7631bebf2df14cf415ae8cc37d27eb2df2fb818.  We have found cases, however, where Meshtastic nodes across the globe can have duplicate Node num as identifiers.

```
"num":115503498,"user":{"id":"!06e2718a","longName":"stevando","shortName":"stdo","macaddr":"6u6hrmY+","hwModel":"RAK4631"
"num":115503498,"user":{"id":"!06e2718a","longName":"AN/ALQ-3dc0","shortName":"3dc0","macaddr":"98NUVT3A","hwModel":"RAK4631"
```

This is a tricky problem to solve as trace routes only include a list of node nums.  We will likely need to add additional logic to associate node with the MeshSense node that reported them.  This adds complexity in the back-end as MeshSense uploaders in the same mesh would add duplicate records for the same node.

Adding the ability to disable rendering routes is a good idea.  We also plan on hiding routes when zoomed out significantly, but have them shown currently while we work out the kinks.

Soltares, 2024-12-18:
Although an id collision was discovered, I found the logic to filter out MQTT was not quite complete.  I am testing some logic changes in development and hope to have a new beta out soon.

Soltares, 2024-12-20:
Version `v1.0.18-beta.2` has been released that should fix the remaining MQTT packets from getting to the global map.  

There is one remaining global route which is the Node num collision (115503498) mentioned above.  Hiding long routes would be a partial solution, but doesn't fix the record from merging properties from both nodes with the same ID.  (Name/Position/etc)

Soltares, 2025-01-22:
With the `v1.0.18` release and global map release, many of the performance issues should now be resolved.  Nodes are identified by a composite key instead of relying on only the Meshtastic ID which resolves more of the cross-world routes.  I suspect any remaining long routes are nodes that are incorrectly reporting position.  Non-MeshSense nodes are also hidden until a certain zoom level to improve performance.

closed 2025-01-22 by Soltares