Affirmatech/MeshSense #44: [feature request] Improved chat and MQTT and others...

Soupbones, 2024-12-20:
Firstly loving Meshsense and the updates! ( can you please add the Beta updates here so we can see whats been fixed? Thats why I'm here today) But a couple of things I would love to see.
1. Using Meshsense for chat messages either on the main Longfast and private channels is a bit cumbersome. It would be great to have a seperate popup window option ( or a way to undock them from the main GUI ) for all chats, and use this window for the conversations.
2.  MQTT so far is not available for the Meshtastic Web client via Bluetooth. It would be great to have the option to see MQTT nodes and conect to MQTT via the app by either Serial, ethernet, Wifi and Bliuetooth. If flooding the mesh with MQTT data is an issue have the option to turn it off so only you see the MQTT data and conversations.
3. Node config. We can see out node settings in Meshsense but it would be great to be able to change and config them in this aop. I think this app should basically become a all round and better alternative to the current Meshtastic Web client, with the view to be mostly for monitoring the local mesh.

Soltares, 2024-12-20:
Thank you for your feedback and suggestions.  I also wondered if and where it made sense to post information on the Beta updates.  The news pop-up in the application is currently used for non-beta releases.  Maybe have a version of that for beta users?  I'm not certain where in GitHub would be best for this information.

To answer your question, the recent beta updates have been focusing on the global map and filtering out MQTT traffic from being sent to the global map.

I agree, a cleaner interface for messaging is needed.  Somewhat related to #26 .

Having MeshSense broker the MQTT connection is an interesting thought and worth thinking about.

Node config is nice, but tricky.  We found that it's not particularly easy to change one setting via the JS API without potentially impacting others.  If all settings in a category aren't provided, they will be cleared or zeroed out.  I agree it would be nice to have though.

Soupbones, 2024-12-20:
Hello Soltares.
Having the app as an MQTT wall between the LoRa Mesh and MQTT would be very handy. It's annoying that the Adroid and IOS apps let all MQTT and LoRa nodes together in one list for you to communicate with, while there is literaly zero options for PCs that can do this.  The whole point of the app is text communications and it seems to usually be at the back with most Meshtastic apps out there. Would be great to see Meshsense as a communications gateway for the user, who like in the Meshtastic android and IOS app can have the best of both worlds. Even an option to have quick access to multiple MQTT servers would be great as well. 
 I get you don't want MQTT traffic spamming out the LoRa network, but at the moment many people are getting into Meshtastic in regions with no other LoRa nodes about, so MQTT is the only choice.
 As for the Beta versions its great to see updates. My software automatically updates, and I recently put in an issue with me not seeing the link lines between nodes on the map on my RAK device. It seems this is working again in the latest Beta. Not sure if this was a fix or not?