Affirmatech/MeshSense #45: What are the black and white node icons with a question mark?

Soupbones, 2024-12-22:
OK, not an issue but this seems a new feature, and I don't know where else to get answers.  But what are the nodes on the map on trace route lines that are black and white and have a question mark next to them? are these just unknown nodes? Or mobile nodes? or unlisted nodes active on the msh in hidden mode? ( you guys need to start a MeshSense Facebook page so people can findo ut this stuff )
[image]

MAWane, 2024-12-22:
It's the fix for #2 showing an estimated location for nodes without GPS or fixed coordinates. Would be a good idea to add this to the FAQ.

N3FJP, 2025-01-03:
We've added an explanation of the black and white node icons with a question mark to the FAQ page.  Thanks for the suggestion!

closed 2025-01-03 by N3FJP