Affirmatech/MeshSense #47: [Feature Request] Implement new traceroute information

Kealper, 2024-12-27:
Hello! As of firmware 2.5, the new traceroute packets give routes both to and from the target node, including SNR values at each hop. It would be awesome if this information was displayed. The relevant JSON snippet from the TRACEROUTE_APP response packet is below:

```json
  "trace": {
    "route": [
      2671296040,
      2086454060,
      56803857
    ],
    "snrTowards": [
      -128,
      -128,
      -69,
      -7
    ],
    "routeBack": [
      56803857,
      1543029502,
      2671296040
    ],
    "snrBack": [
      -47,
      -73,
      -22,
      21
    ]
  }
```

Explanation of packet structure: To keep compatibility with older firmware, the new format reuses the old `route` array to trace the route towards the target, but adds new `snrTowards`, `routeBack`, and `snrBack` arrays which are ignored by older firmware.

`routeBack` is an array that's a reverse of `route`'s so the first node in the array is the target with the last node is the closest to the source. Since the route to and from a target can be different, this can't always be assumed to be the same as `route` and in my experience, it's common for it to be slightly different.

`snrTowards` and `snrBack` are SNR values ranging from +31.75dB to -32dB in with 0.25dB resolution packed into a signed 8-bit int. To get the human-readable version of the SNR for each hop, just divide the value by 4. The final value in each array is the SNR for the target (`snrTowards`) or the source (`snrBack`) and the other values in the arrays correspond to to the node numbers in the same positions of their respective arrays, e.g. element 2 of `snrTowards` is the SNR that the node in element 2 of `route` received the traceroute packet at from the previous node in the route. The value -128 is a special value that gets assigned by a node when the previous node in the route did not put SNR information, which happens when that previous node was running older firmware that doesn't understand the new format.

### Proposal

I feel like it would be a large change to show both towards and reverse routes on the map so I think the way it displays it now with only showing the routes towards would be fine to keep in place if that part would be too much work. I think a good place to show the detailed traceroute information would be in the on-hover tooltip in the node list. Currently, it looks like this for the traceroute packet above:

```
Traceroute KlpB -> Hkry -> ka0s -> SigR -> KlpC
```

I think it would be nice to change that to something like this, though the exact formatting could be different:

```
Traceroute:
KlpB --[?dB]--> Hkry --[?dB]--> ka0s --[-17.25dB]--> SigR --[-1.75dB]--> KlpC
KlpB <--[5.25dB]-- Hkry <--[-5.5dB]-- MCF1 <--[-18.25dB]-- SigR <--[-11.75dB]-- KlpC
```

The node `ka0s` in that traceroute is on an older firmware so it does not pass along SNR information for the hops before it, but all other nodes are running some version of firmware 2.5 so they pass along their received SNRs. Having SNR information for the hops could also be used to color the route lines on the map with higher SNRs being more green, moderate SNRs fading to more yellow, and terrible SNRs fading more towards red. The special -32dB SNR could color the line the color it is now (or something like gray) to denote that the hop's SNR was not known.

Soltares, 2025-01-03:
Thank you very much for taking the time to write a beautifully detailed feature request. :+1:   It is greatly appreciated and I will work to add the SNR reporting as you propose into MeshSense.

gafu42, 2025-08-02:
There are different snr values depending on meshtastic node chip.
The older sx1278/76 deliver values up to ~+13dB, the newer sx1262 max out at +6dB snr.
Lowest values are -20dB in reality at both hardwares.

to me it looks like the values itself are comparable, only the newer ones do not go higher as +6dB

Maybe this is helpful when implementing some visuals.

Kealper, 2025-08-02:
To throw a spanner into the works on that, the T1000-E has an SX1262 and reports SNRs of +10dB to +12dB and I think that's down to it having basically no feedline between the antenna and the radio module.