Affirmatech/MeshSense #49: [Feature Request] Options for setting a hard-maximum precision for global MeshSense Map

Kealper, 2024-12-30:
Hello! The new global map is really cool!

Something I would love to have is an option in the section for enabling data collection is the ability to define a maximum precision for device locations that are reported to the global map. For example: If a person's node is reporting its position at a full 32 bits but my MeshSense instance is configured to allow a maximum of 16 bits of precision with this proposed setting, the person's node position is reduced to 16 bits. If they were already reporting something less-precise, such as 13 bits, then that information is shared with the global map as-is.

The reason I wish something like this was added is that there's a general push in Meshtastic's development to be more privacy-first as far as location information goes. There's already been options to let users tell other nodes that they don't wish to have their location information published to the "main" MQTT instance, which is enabled by default. I love that I can submit information to the global map from my "perspective" in my area's mesh, but I do wish that, on the occasion that someone sets their position to the full 32 bits, they don't inadvertently have the room in their house that they're currently sitting in uploaded online for the world to see. At the very least, at least not uploaded by something that I personally was running.

Soltares, 2025-01-02:
Thank you for the suggestion!  That's an interesting idea and worth considering.  My immediate concern with overriding a reported position by the MeshSense instance uploading to the global map, is what would happen when a different MeshSense uploader nearby then reports the 32-bit position to the map.  The node could potentially jump between the two locations depending on which update comes through last.

The other side of the coin is that Meshtastic users do have control over the precision they broadcast and whether or not they broadcast position inside of Device settings from the Meshtastic app, so control ultimately resides with them and how their devices are configured.  Granted, I agree that one inadvertent upload probably shouldn't persist forever.  The global map is currently configured to purge node data from the system if there has been no update for 12 hours.

Kealper, 2025-01-04:
That's a good point, I hadn't thought about how to handle conflicts with multiple MeshSense instances on a mesh uploading different amounts of precision.

I had written up, then backspaced, a thing about having a fixed amount of precision loss unless a user had the OkToMQTT flag set in their LoRa settings, as that setting is effectively an "allow my position to be uploaded to the public Internet or not" which is set to off by default but the more I thought about that, the more I realized it doesn't quite help the problem you've mentioned. I'd guess the way to handle that would just be to accept the most-precise location but getting that to be handled correctly means doing extra legwork on the server that's collecting the data to determine if the value it just got was a "better" position than the previous ones or not. Oof.

The original intent of the feature request was to have a setting that allowed MeshSense operators to optionally hide users' exact positions if they either didn't know their positions were being shared on the Internet or explicitly didn't want to share that information to anyone other than the local RF mesh. I think the best solution may just be the solution that's already available, which is that users can control how accurate their position is and if they do unintentionally send a more accurate position than they wanted to, it'll be gone off of the MeshSense map in less than a day at worst.