Affirmatech/MeshSense #50: Adding power metrics to MeshSense

lupusworax, 2024-12-30:
After Telemetry of environmental sensors were added lately it would be amazing if power metrics could be added as well. Especially for solar nodes where a solar power manage board directly feeds 5 volt to the node its hard to determine the actual charge of the solar management  connected batteries, the I2C used sensor does not show up yet on the metrics next to temp, humidity, air quality etc. would be awesome to see that so you see upfront hows the voltage and power draw of the actual battery pack connected to the solar charging management board.

Soltares, 2025-01-02:
Thanks for the suggestion!  Do the additional values you're referring to appear in the environmentMetrics section of the node in question?  If so, can you please send me an example output?

```
  "environmentMetrics": {
    "temperature": 20.017494201660156,
    "lux": 4,
    ....
  },
```

Is it `power` and `current` as listed in the [API docs](https://js.meshtastic.org/types/Protobuf.Telemetry.EnvironmentMetrics.html)?

lupusworax, 2025-01-03:
Oh I see its already there ( all down on the bottom) , probably just not broken out on the overview on the main page: 
{
  "num": 306791784,
  "user": {
    "id": "!12494568",
    "longName": "Solar Node  4568",
    "shortName": "🌞",
    "macaddr": "09USSUVo",
    "hwModel": "NRF52_PROMICRO_DIY",
    "isLicensed": false,
    "role": "CLIENT",
    "publicKey": "oWGZHC2omK4ZAVdFQl8T8MYZ40NcoASnY7En7QAZZXE="
  },
  "position": {
    "time": 0,
    "locationSource": "LOC_UNSET",
    "altitudeSource": "ALT_UNSET",
    "timestamp": 0,
    "timestampMillisAdjust": 0,
    "PDOP": 0,
    "HDOP": 0,
    "VDOP": 0,
    "gpsAccuracy": 0,
    "fixQuality": 0,
    "fixType": 0,
    "satsInView": 0,
    "sensorId": 0,
    "nextUpdate": 0,
    "seqNumber": 0,
    "precisionBits": 0
  },
  "snr": 4,
  "lastHeard": 1735902520.305,
  "deviceMetrics": {
    "batteryLevel": 101,
    "voltage": 5.1020002365112305,
    "channelUtilization": 3.8899998664855957,
    "airUtilTx": 0.4383055567741394,
    "uptimeSeconds": 203111
  },
  "channel": 0,
  "viaMqtt": false,
  "hopsAway": 0,
  "isFavorite": false,
  "rssi": -75,
  "environmentMetrics": {
    "temperature": -0.11999999731779099,
    "relativeHumidity": 37.2841796875,
    "barometricPressure": 966.6107788085938,
    "voltage": 3.996000051498413,
    "current": -5.099999904632568
  },
  "trace": null,
  "powerMetrics": {
    "ch3Voltage": 3.996000051498413,
    "ch3Current": -5.599999904632568
  }

Soltares, 2025-01-03:
I was looking to implement this, however I noticed that `environmentMetrics.voltage` is deprecated in favor of `powerMetrics`.

> To be depreciated in favor of PowerMetrics in Meshtastic 3.x

`powerMetrics`, however, has three different channels for reporting voltage and current:

https://js.meshtastic.org/types/Protobuf.Telemetry.PowerMetrics.html

Which channel should be displayed?  Or should I use the first channel that has a value?

lupusworax, 2025-01-03:
In my  case its channel 3 but I cant say for sure if its the same for all?  probably has something to do with the I2C address?!
[image]

HougeDK, 2025-02-01:
If you use a INA3221, there are 3 channels in the powerMetrics data.
So i think it should support 3 channels

lupusworax, 2025-02-01:
In my example I am using an INA219

HougeDK, 2025-02-01:
This is with a INA3221

[image]