Affirmatech/MeshSense #59: MeshSense fails to open on Ubuntu 22.04 since 13th of Feb

willp618, 2025-02-18:
since 13th of Feb meshsense no longer opens on Ubuntu 22.04 

this is the error 

Error: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /tmp/.mount_meshseKBWw3b/resources/app.asar.unpacked/resources/api/simpleble.node)
    at process.func [as dlopen] (node:electron/js2c/node_init:2:2559)
    at Module._extensions..node (node:internal/modules/cjs/loader:1602:18)
    at Object.func [as .node] (node:electron/js2c/node_init:2:2559)
    at Module.load (node:internal/modules/cjs/loader:1295:32)
    at Module._load (node:internal/modules/cjs/loader:1111:12)
    at c._load (node:electron/js2c/node_init:2:16955)
    at Module.require (node:internal/modules/cjs/loader:1318:19)
    at require (node:internal/modules/helpers:179:18)
    at Object.<anonymous> (/tmp/.mount_meshseKBWw3b/resources/app.asar.unpacked/resources/api/index.cjs:84722:30)
    at Module._compile (node:internal/modules/cjs/loader:1484:14) {
  code: 'ERR_DLOPEN_FAILED'

this is an issue as i not able to update this machine to Ubuntu 24.04 which a lot of answers from a google search say to do 
i tried to update to the needed GLIBCXX but it resulted in my Ubuntu machine black screening and no longer booting thankfully i could restore it from a backup 

I'm not the only one with this issue a local on my mesh was running meshsense on a Debian 12 machine and he made the mistake of clicking the update button which killed his machine he also manage restore from a backup 

IMO there needs to be support for older but still supported OSs even the current download on the website doesn't work with same error on both Ubuntu 22.04 and Debian 12 

I am missing using meshsense as the web client doesn't always work for me

syssi, 2025-02-18:
+1 

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

```
$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_3.4.29
GLIBCXX_3.4.30
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

Soltares, 2025-02-18:
Thank you for the report!  I have rebuilt MeshSense v1.0.18 on an Ubuntu Jammy base OS and this should resolve the glibc dependency error.  I was previously building on Arch Linux and it seems it is better to use an LTS system for better compatibility.  Please let me know if it works after redownloading from the website.

syssi, 2025-02-18:
It's working again! Thanks for the quick response and solution!

Soltares, 2025-02-18:
Fantastic!!  Thanks also for the quick confirmation! 😄

closed 2025-02-18 by Soltares

willp618, 2025-02-19:
Can confirm this is working for me as well thanks for the quick response and fix