Affirmatech/MeshSense #60: Memory Leak in MeshSense Helper (GPU)

mblauser, 2025-02-20:
build: 1.0.18
OS: MacOS Sonoma 14.7.2 using Apple Silicon M2 Pro

I've had version 1.0.18 of MeshSense running for 65 days, and MeshSense Helper (GPU) slowly consumes an increasing amount of memory. It is now up to 19.48 GB of reported memory use, 15 threads, 194 ports.

[image]

A deeper dive reveals:
```
  Real Memory Size: 113.7 MB
  Virtual Memory Size: 443.02 GB
  Shared Memory Size: 119.6 MB
  Private Memory Size: 43.5 MB
```

[image]

[image]

What may I provide (and how) to help further diagnose or reveal the source of the memory leak?

Soltares, 2025-02-20:
Thanks for the report!  Are you running the version from the website or a built version?  If you built it yourself, which version of Node are you running?

I have seen mention in meshtastic/js of a possible Memory Leak condition for node versions under version 20

https://github.com/meshtastic/js/issues/124

mblauser, 2025-02-24:
Downloaded an earlier version from the website; using the built-in upgrade mechanism to stay current up to v1.0.18.

Currently have Node v23.7.0 installed, although it gets updated via homebrew frequently. I do not know which version of Node was used when launching this current instance of MeshSense 2+ months ago.