Affirmatech/MeshSense #61: Headless flag crashes

Famicoman, 2025-02-25:
Hello, trying to run with the `--headless` flag on a server and seeing this in the logs. Looks like the app crashes trying to create a ` GtkStyleContext`:

```
[1:0225/032956.542524:ERROR:bus.cc(407)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
[1:0225/032956.987051:ERROR:bus.cc(407)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
[1:0225/032956.987134:ERROR:bus.cc(407)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
[1:0225/032957.018573:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1:0225/032957.018835:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1:0225/032957.018939:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1:0225/032957.018951:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1:0225/032957.018958:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
DIRNAME /tmp/.mount_meshse5e209D/resources/app.asar/out/main
API_PATH /tmp/.mount_meshse5e209D/resources/app.asar.unpacked/resources/api/index.cjs
[electron] Arguments [ '/tmp/.mount_meshse5e209D/meshsense', '--headless', '--no-sandbox' ]
[electron] Checking for updates on channel null
Checking for update
[32:0225/032957.193904:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[32:0225/032957.197820:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[32:0225/032957.197914:ERROR:gl_display.cc(767)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
[32:0225/032957.198084:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[32:0225/032957.198215:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[32:0225/032957.198306:ERROR:gl_display.cc(767)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[32:0225/032957.198341:ERROR:gl_display.cc(801)] Initialization of all EGL display types failed.
[32:0225/032957.198401:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[32:0225/032957.210081:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[32:0225/032957.210153:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[32:0225/032957.210181:ERROR:gl_display.cc(767)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
[32:0225/032957.210338:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[32:0225/032957.210375:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[32:0225/032957.210428:ERROR:gl_display.cc(767)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[32:0225/032957.210483:ERROR:gl_display.cc(801)] Initialization of all EGL display types failed.
[32:0225/032957.210556:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[32:0225/032957.220666:ERROR:viz_main_impl.cc(181)] Exiting GPU process due to errors during initialization
Generated new staging user ID: 32952e82-add7-5def-9bff-45f4756eccf5
[47:0225/032957.268606:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[47:0225/032957.268750:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[47:0225/032957.268830:ERROR:gl_display.cc(767)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
[47:0225/032957.268920:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[47:0225/032957.269021:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[47:0225/032957.269070:ERROR:gl_display.cc(767)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[47:0225/032957.269122:ERROR:gl_display.cc(801)] Initialization of all EGL display types failed.
[47:0225/032957.269163:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[47:0225/032957.283663:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[47:0225/032957.283757:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[47:0225/032957.283889:ERROR:gl_display.cc(767)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
[47:0225/032957.284004:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[47:0225/032957.284116:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[47:0225/032957.284238:ERROR:gl_display.cc(767)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[47:0225/032957.284335:ERROR:gl_display.cc(801)] Initialization of all EGL display types failed.
[47:0225/032957.284416:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[47:0225/032957.285430:ERROR:viz_main_impl.cc(181)] Exiting GPU process due to errors during initialization
[55:0225/032957.335308:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[55:0225/032957.335440:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[55:0225/032957.335638:ERROR:gl_display.cc(767)] eglInitialize OpenGL failed with error EGL_NOT_INITIALIZED, trying next display type
[55:0225/032957.335757:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
ERR: Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[55:0225/032957.335850:ERROR:gl_display.cc(497)] EGL Driver message (Critical) eglInitialize: Could not open the default X display.
[55:0225/032957.335928:ERROR:gl_display.cc(767)] eglInitialize OpenGLES failed with error EGL_NOT_INITIALIZED
[55:0225/032957.336061:ERROR:gl_display.cc(801)] Initialization of all EGL display types failed.
[55:0225/032957.336181:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[55:0225/032957.337651:ERROR:angle_platform_impl.cc(44)] Display.cpp:1083 (initialize): ANGLE Display::initialize error 12289: Could not open the default X display.
[... cut ...]
[bluetooth] Begin Scanning
[api to electron] { event: 'setUpdateChannel', body: undefined }
[electron] Set update channel undefined
[news] Checking for news
Server listening { address: '::', family: 'IPv6', port: 5920 }
CREATING WINDOW
[1:0225/032957.817145:ERROR:bus.cc(407)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")

(meshsense:1): Gtk-ERROR **: 03:29:57.817: Can't create a GtkStyleContext without a display connection
```

bplein, 2025-03-23:
I am seeing the same thing on a headless Ubuntu server. 

My guess is that this has only been working on systems with a GUI installed (even if not used). It's not a truly headless implementation.

It would be great to hear from the developer.

Soltares, 2025-03-24:
Thanks for the report.  A few things worth sharing:

- `v1.0.18` also requires the `HEADLESS=1` environment variable to be set.  The current `v1.1.0` beta removes this requirement.

- In [another comment](https://github.com/Affirmatech/MeshSense/issues/22#issuecomment-2689220263), folks have had success expanding the invocation to the following:

```
dbus-run-session xvfb-run ./meshsense-beta-arm64.AppImage --headless --disable-gpu --in-process-gpu --disable-software-rasterizer
```

- I agree in that it seems electron is still trying to pull some GUI libraries to start even if they are not used.  I believe the required libraries include the following:

```
sudo apt install libatk-bridge2.0-0
sudo apt install libcups2
sudo apt install libgdk-pixbuf-2.0-0
sudo apt install libgtk-3-0
sudo apt install libgbm1
sudo apt install xvfb
```

This is not ideal, but I don't have a great answer to this problem.  Electron brings the following benefits:

- Easy distribution and execution (single executable/installable)
- Automatic updates and package signing via [electronBuilder](https://www.electron.build/auto-update.html)
- Bundled Node runtime at a consistent version

If there is an alternative I'm open to trying it, but I'm not sure what that would be.

If I were to ship the back-end without electron we lose these benefits and I'm nervous that the overhead of manual updates, Node version requirements, and support questions would outweigh the benefits.  Although it hurts me a little inside, the quickest path to success right now is to install those packages.