Affirmatech/MeshSense #64: Please increase `globalTracerouteRateLimitSec` and `tracerouteRateLimit`

GUVWAF, 2025-02-27:
Due to the excessive use of traceroutes by MeshSense, people are requesting higher rate-limits on traceroutes in the firmware: https://github.com/meshtastic/firmware/issues/6173. 
Since this would also affect manual traceroutes, which I believe would be a shame, I would like to ask to increase `globalTracerouteRateLimitSec` and `tracerouteRateLimit` (per node), e.g. to 3 min. and 3 hours respectively.

RCGV1, 2025-02-27:
MeshSense's default traceroute interval is set to 15 minutes per node. In a large mesh network with 100+ nodes, this results in traceroutes being sent as frequently as every 15-30 seconds because packets follow different paths due to network redundancy.  Hop count changes packet to packet, naturally. 

Suggested Improvements:

Increase the default traceroute interval to one hour to reduce unnecessary requests.
Implement logic to limit redundant traceroutes when different paths are detected within a short time window. Introduce a watchdog timer to prevent high-frequency tracerouting from running indefinitely in the background without user intervention. Due to the nature of mesh networks, where redundancy ensures dynamic path selection, the current behavior leads to excessive traceroute activity, degrading network performance.

Soltares, 2025-02-27:
I will absolutely take a look and help how I can.  A global rate-limit queue was implemented previously in #18.  I'm not at my desk currently, but I will take some time to write up the current logic on trace routes and we can see how to best adjust.  Thanks!

mblauser, 2025-02-27:
> excessive use of traceroutes

^^ That is subjective

I'd argue that the reported issue only affects congested mesh networks and that additional traceroute limits would unnecessarily burden less-congested mesh networks. However, I do agree that the current (and any additional) rate limits placed on manual traceroutes are unfortunate.

### Proposed Compromises

I propose an alternative compromise:  Dynamically impose traceroute rate limits within MeshSense based on current levels of mesh congestion.  We might set limits globally and/or on individual queued traceroutes based on remote nodes' reported congestion values. 

We might also consider increasing default global limits, but allow for MeshSense operators to manually reduce configurable limits when appropriate.  This would reduce the frequency of traceroutes on the majority of MeshSense instances.

Furthermore, I propose implementing node data and traceroute caching (of an appropriate staleness) within MeshSense, to enable preservation of recent traceroute results across application restarts and version upgrades. The majority of rapid traceroutes occur when the application is first launched as it collects initial node data. 

### Better Firmware Strategy

At its core, traceroutes, as currently implemented in the meshtastic firmware, are network amplification attack vectors.  Remote nodes may attempt to play nice, but ultimately the responsibility resides on participating mesh nodes to decide whether to reply.

I would prefer that firmware developers consider implementing improved traceroute efficiency, or less-costly transmission propagation targeting (preferring) last-known routes.  Alternatively, find other areas to reduce congestion; e.g. less-frequent battery, temperature, and location packets if values have not changed significantly.

RCGV1, 2025-02-27:
Even with a not already congested mesh network mesh sense will still attempt to send a large amount of trace routes. The limits should be based on node counts and not channel utilization. I would also like to have the shorter trace route limits to be only enabled if the app is opened and not in the background to only allow people to send a lot of trace routes if they are actively monitoring the app.

GUVWAF, 2025-02-27:
> I'd argue that the reported issue only affects congested mesh networks and that additional traceroute limits would unnecessarily burden less-congested mesh networks.

Requesting a traceroute to *every* node *every* 15 min. is way higher than any automatic packet transmission in the firmware.

> I would prefer that firmware developers consider implementing improved traceroute efficiency, or less-costly transmission propagation targeting (preferring) last-known routes

The 2.6 pre-release of the firmware will use more efficient routing for packet directed to one specific node, like traceroutes. 

>  Alternatively, find other areas to reduce congestion; e.g. less-frequent battery, temperature, 

This has been reduced in recent firmware already, and can be manually controlled.

> and location packets if values have not changed significantly.

When using the "smart position" option (enabled by default) location is already only sent when you're moving.

Soltares, 2025-02-27:
There is a bit of misinformation and I would like to try and paint a clearer picture of the current trace route strategy.  I've reviewed the current logic and can confirm that ALL of the following conditions must be met for a trace route to be requested:

**Trigger condition:** An incoming packet is received from Node X.  @mblauser  This is separate from the node list being populated on startup, however some queued packets on the device are delivered in quick succession on startup.  In other words, not all nodes will have trace routes requested on startup, only nodes that have had packets delivered to MeshSense.

- Node X is not a direct neighbor with 0 hops
- Automatic trace routes are enabled in settings (default)
- Node X has a known position
- Node X does not have trace route data **OR** Node X hop count has changed
- Node X has not had a trace request within `tracerouteRateLimit` minutes (currently 15m, proposed change to 60m)

The request is then queued in a global rate limiter which will request trace routes every `globalTracerouteRateLimitSec` (currently 30 seconds)

To reiterate -- trace routes are not triggered by time every 15 minutes.  Only when the above logic allows a trace route to execute.  I am okay with adjusting the per-node limit with 60 minutes by default and/or dynamically adjusting this based on count of active nodes.  The 15 minute default is already adjustable by the user in the settings screen.

[image]

GUVWAF, 2025-02-27:
Thanks for the clarification, those are good trigger conditions already. Still, in the worst case it would request a traceroute to every active node every 15 min. Hop counts change constantly, since (at least pre-2.6) there are no predefined routes.

I would be happy to see an increase in the default per-node limit (and also a minimum restriction), and preferably also the global limit.

N3FJP, 2025-02-27:
Hi All,

Thanks for sharing your thoughts.  I could not verify the premise that MeshSense is responsible for a large portion of network traffic.  In fact, in our local mesh, the packet traffic MeshSense induces is minimal.  Here is the quick analysis I did that you can repeat:

-  At the bottom of the main MeshSense form, click the CSV button.
-  Press Ctrl + C to copy the data.
-  Start Notepad and paste the data.
-  Save the file with a CSV extension.
-  Open the file with a spreadsheet such as Excel.
-  Sort the records on type.
-  Count each packet type.

Our local mesh currently has 23 active nodes.  In the last 3 hours we have had 320 packets.  Of those 320 packets, 0 have been RouteRequest packets and 0 have been RouteDiscovery packets.
  
[image]

Obviously we do occasionally have RouteRequest and RouteDiscovery packets as hops do sometimes change, so I am not suggesting that it is always 0, but it is clear that they represent a very tiny portion of our mesh traffic.

I encourage you to repeat this analysis for your cluster.  We will be eager to hear what you learn!

Thanks again!

GUVWAF, 2025-02-27:
It depends on the settings, that's why I would also vote for having a minimum interval.
This was a report we got on Discord: 

[image]

RCGV1, 2025-02-27:
> Hi All,
>
> Thanks for sharing your thoughts. I could not verify the premise that MeshSense is responsible for a large portion of network traffic. In fact, in our local mesh, the packet traffic [...]

Our mesh has 2-3 redundant paths available to a majority of users.  We have 132 devices and you will see packets from every node about 1-3 times per hour after accounting for nodeID, telemetry, location packets, and occasionally text messages ... which means a new packet from some node every 10-30seconds even.  Its worse during traceroutes which are floods... causing a cascade of meshsense responding to meshsense.  Our redundant paths and well connected mesh results in hop counts changing for almost every packet you see from a particular specific node.  You may send me a packet 5 times in a row and get 3 different hop counts.

mblauser, 2025-02-28:
Personally I run MeshSense at 40 minutes `tracerouteRateLimit` maximum per node.

Here is a 2025-02-27 breakdown of the past 654 transmissions in Las Vegas among 75 observed nodes recently:

type | count | pct
-- | -- | --
POSITION_APP | 258 | 39%
TELEMETRY_APP | 231 | 35%
encrypted | 113 | 17%
NODEINFO_APP | 48 | 7%
undefined | 3 | 0%
TRACEROUTE | 1 | 0%
  | 654 | 100%

[image]

> This was a report we got on Discord:
> 
> [image]

I see correlation, but not necessarily causation.  I've observed multiple instances of high channel saturation for 2-3 days in regional pockets of Las Vegas, and then suddenly back to normal for the next several days, without any changes in MeshSense. I suspect client nodes not configured as CLIENT_MUTE when best suited, radio profiles not using LONG_FAST, ATAK, other LoRa transmissions, or other 915 MHz radio devices operating in the area.

> Unless anyone feels otherwise, I will plan to increase default `tracerouteRateLimit ` from 15 minutes to 60 minutes (with a 15 minute minimum) and double the `globalTracerouteRateLimitSec` from 30 to 60 seconds.

I'm in favor of increasing the default `tracerouteRateLimit` to 45 minutes with a 10 minute minimum, but not increasing the (default) `globalTracerouteRateLimitSec` of 30 seconds unless given the ability to lower the value in the settings.  

I have not seen convincing evidence to conclude that MeshSense is causing congestion or undue stress; especially not when `tracerouteRateLimit` is set to 30-40 minutes max per node, and `globalTracerouteRateLimitSec` is at the firmware limit of 30 seconds.

> [@mblauser](https://github.com/mblauser) I wonder if we revisit [#10](https://github.com/Affirmatech/MeshSense/pull/10) to have better visibility into network congestion?

I would very much like to see this, but being that the UI changes would affect multiple files and my limited CSS skills, I have been apprehensive to pick up where the proposed code changes currently stand. I welcome your expertise.

Talie5in, 2025-02-28:
> I have not seen convincing evidence to conclude that MeshSense is causing congestion or undue stress; especially not when `tracerouteRateLimit` is set to 30-40 minutes max per node, and `globalTracerouteRateLimitSec` is at the firmware limit of 30 seconds.

Between this comment, and the earlier one with a list of conditions required...

How does that change when there are 10 MeshSense users all running automatic traceroute?
Does MeshSense detect the other traceroutes (any, not just meshsenses as that's likely not possible) and backoff for NodeX (Destination)?

N3FJP, 2025-02-28:
Hi All,

We will be happy to modify MeshSense to minimize network traffic, but we need to be sure of the cause and magnitude, if indeed it is related to MeshSense.  We've been unable to duplicate any traffic issues related to MeshSense on our 32 node mesh here.

From the two analyses performed so far, traceroutes have not been identified as adding any significant overhead.  Anecdotal evidence from other meshes noticing high network traffic while running MeshSense certainly is worthy of more investigation, but does not provide enough detail to conclude traceroutes are the cause, or if it is MeshSense related at all.

Traceroute requests sent out by MeshSense would indeed have been counted and identified as RouteRequest packets, had they occurred.  There were 0 occurrences during my 3 hour analysis of 320 packets (detailed above).

MeshSense does not routinely send out other transmissions, aside from actions initiated by the user, such as messages.

My node running MeshSense, which is a central hub for our local cluster, is reporting these values:

"channelUtilization": 4.106666564941406,
 "airUtilTx": 1.2401944398880005,

According to Perplexity "In the context of Meshtastic networks, which are designed for low-bandwidth, long-range communication, these values represent light usage. The network has plenty of capacity for additional traffic without risking congestion or significant delays."

Rather than guessing, let's quantify with certainty if MeshSense is the cause and if so, what specifically is driving the traffic.  Please have users with high traffic clusters perform the analysis steps detailed above, so we can definitively identify the cause and magnitude of any traffic MeshSense may be inducing.

Thanks for working with us on this!

nullrouten0, 2025-02-28:
Our mesh is avg 10% chUtil until a MeshSense traceroute storm kicks up.   Nodes start tracerouteing each other because hop count is constantly changing packet to packet.  Here's the DE-DUPLICATED view of traceroute packets.  You can see wild wild swings in load.  Humans dont do this.  This is <200 nodes and trace routes in the middle of the night too.  (times in UTC). Our ChUtil is nominally 10% and during a storm it's 40%. 

sqlite> SELECT strftime('%Y-%m-%d %H:00', import_time) AS hour, COUNT(*) AS packet_count
   ...> FROM traceroute
   ...> GROUP BY hour
   ...> ORDER BY hour;
+------------------+--------------+
|       hour       | packet_count |
+------------------+--------------+
| 2025-02-24 06:00 | 384          |
| 2025-02-24 07:00 | 596          |
| 2025-02-24 08:00 | 183          |
| 2025-02-24 09:00 | 16           |
| 2025-02-24 10:00 | 12           |
| 2025-02-24 11:00 | 46           |
| 2025-02-24 12:00 | 17           |
| 2025-02-24 13:00 | 124          |
| 2025-02-24 14:00 | 107          |
| 2025-02-24 15:00 | 86           |
| 2025-02-24 16:00 | 136          |
| 2025-02-24 17:00 | 144          |
| 2025-02-24 18:00 | 61           |
| 2025-02-24 19:00 | 374          |
| 2025-02-24 20:00 | 152          |
| 2025-02-24 21:00 | 311          |
| 2025-02-24 22:00 | 242          |
| 2025-02-24 23:00 | 428          |
| 2025-02-25 00:00 | 126          |
| 2025-02-25 01:00 | 193          |
| 2025-02-25 02:00 | 108          |
| 2025-02-25 03:00 | 59           |
| 2025-02-25 04:00 | 6            |
| 2025-02-25 05:00 | 70           |
| 2025-02-25 06:00 | 37           |
| 2025-02-25 07:00 | 54           |
| 2025-02-25 08:00 | 60           |
| 2025-02-25 09:00 | 12           |
| 2025-02-25 10:00 | 49           |
| 2025-02-25 11:00 | 18           |
| 2025-02-25 12:00 | 41           |
| 2025-02-25 13:00 | 49           |
| 2025-02-25 14:00 | 44           |
| 2025-02-25 15:00 | 111          |
| 2025-02-25 16:00 | 92           |
| 2025-02-25 17:00 | 62           |
| 2025-02-25 18:00 | 24           |
| 2025-02-25 19:00 | 5            |
| 2025-02-25 20:00 | 6            |
| 2025-02-25 21:00 | 15           |
| 2025-02-25 22:00 | 32           |
| 2025-02-25 23:00 | 23           |
| 2025-02-26 00:00 | 55           |
| 2025-02-26 01:00 | 59           |
| 2025-02-26 02:00 | 325          |
| 2025-02-26 03:00 | 198          |
| 2025-02-26 04:00 | 18           |
| 2025-02-26 05:00 | 32           |
| 2025-02-26 06:00 | 54           |
| 2025-02-26 07:00 | 139          |
| 2025-02-26 08:00 | 60           |
| 2025-02-26 09:00 | 7            |
| 2025-02-26 10:00 | 13           |
| 2025-02-26 11:00 | 7            |
| 2025-02-26 12:00 | 7            |
| 2025-02-26 13:00 | 7            |
| 2025-02-26 14:00 | 51           |
| 2025-02-26 15:00 | 72           |
| 2025-02-26 16:00 | 73           |
| 2025-02-26 17:00 | 167          |
| 2025-02-26 18:00 | 71           |
| 2025-02-26 19:00 | 65           |
| 2025-02-26 20:00 | 232          |
| 2025-02-26 21:00 | 88           |
| 2025-02-26 22:00 | 138          |
| 2025-02-26 23:00 | 138          |
| 2025-02-27 00:00 | 392          |
| 2025-02-27 01:00 | 333          |
| 2025-02-27 02:00 | 107          |
| 2025-02-27 03:00 | 274          |
| 2025-02-27 04:00 | 150          |
| 2025-02-27 05:00 | 220          |
| 2025-02-27 06:00 | 308          |
| 2025-02-27 07:00 | 319          |
| 2025-02-27 08:00 | 64           |
| 2025-02-27 09:00 | 65           |
| 2025-02-27 10:00 | 71           |
| 2025-02-27 11:00 | 90           |
| 2025-02-27 12:00 | 110          |
| 2025-02-27 13:00 | 189          |
| 2025-02-27 14:00 | 57           |
| 2025-02-27 15:00 | 53           |
| 2025-02-27 16:00 | 77           |
| 2025-02-27 17:00 | 42           |
| 2025-02-27 18:00 | 376          |
| 2025-02-27 19:00 | 154          |
| 2025-02-27 20:00 | 106          |
| 2025-02-27 21:00 | 104          |
| 2025-02-27 22:00 | 149          |
| 2025-02-27 23:00 | 168          |
| 2025-02-28 00:00 | 215          |
| 2025-02-28 01:00 | 86           |
| 2025-02-28 02:00 | 360          |
| 2025-02-28 03:00 | 217          |
| 2025-02-28 04:00 | 198          |
| 2025-02-28 05:00 | 43           |
| 2025-02-28 06:00 | 278          |
| 2025-02-28 07:00 | 576          |
| 2025-02-28 08:00 | 3            |
| 2025-02-28 11:00 | 72           |
| 2025-02-28 12:00 | 47           |
| 2025-02-28 13:00 | 24           |
| 2025-02-28 14:00 | 115          |
| 2025-02-28 15:00 | 157          |
| 2025-02-28 16:00 | 25           |
+------------------+--------------+

nullrouten0, 2025-02-28:
This might be exacerbated by a very technically savvy community here in Silicon Valley, and I assume over half of our users have run MeshSense at some point.  Here are some examples of User A tracing B, then B tracing back A immediately. 

[image]

[image]

prokrypt, 2025-02-28:
Just a note: the airtime/packet stats from a node you're sending out traceroutes from doesn't show the whole picture. The traceroutes you send out are small, and they get larger with each hop, taking more and more airtime. If you send a traceroutes to a neighbor, the packet that comes back will be pretty small, but you have to remember it's still propagating and growing up to the hop limit.

N3FJP, 2025-02-28:
Thanks guys, this is helpful.

If you go here, you can see who in your cluster is running MeshSense right now (larger purple icons that remain displayed on the map when you zoom out):

https://meshsense.affirmatech.com/

Are you 100% sure the "storms" are mostly traceroutes?  Since all nodes transmit relatively frequently anyway, I'm not clear why one traceroute request would induce more TRs?  When I send out a TR here, it doesn't generate more in our 32 mesh cluster.

We would really appreciate seeing actual counts of the packet types during these storms from multiple clusters.  It only takes a few minutes to perform the analysis with the tools MeshSense provides.  Please see above for the easy steps.

It still seems like there might be something more at play here than traceroutes.  If you get a chance to report a packet type count, that would be extremely helpful, not only ensuring us that we are on the right track with our TR suspicions, but it would be something that we can measure against after we make changes to MeshSense and release the new version to assess improvement.  MeshSense TR behavior changes aren't going to make any difference for our local cluster, so we really need more detail from those experiencing these storms to determine the effectiveness of our modifications.

Thanks all!

Soltares, 2025-02-28:
Thank you all for the information and feedback.  One thing nobody wants is a congested mesh network.  We can certainly continue to research to ensure that MeshSense isn't causing undue stress on a network.  Changing the defaults, however, is easy and will at the very least attempt to take advantage of low hanging fruit to help networks experiencing high channel utilization.

For the sake of providing relief to congested networks, I'm inclined for now to err on the side of caution and set:

- `tracerouteRateLimit` from 15 minutes to 60 minutes (adjustable to 15 minutes minimum)
- `globalTracerouteRateLimitSec` from 30 seconds to 60 seconds (not currently user adjustable, but perhaps in the future)

I am going to go ahead and make this change and publish beta version `1.1.0.-beta.2` which will automatically update for those that have subscribed to the beta release channel.

---

**A few notes:**

This will be the first published beta since updating the Meshtastic JS API and Protobufs. (Hence the minor version increase)  In particular, packet type naming has changed in addition to some payload formatting.

|Old|New|Comment|
|--|--|--|
|TRACEROUTE|RouteRequest|@GUVWAF - This is an artificial packet log entry by MeshSense to track transmitted trace route requests|
|TRACEROUTE_APP|RouteResponse||
|POSITION_APP|Position||
|TELEMETRY_APP|DeviceMetrics, PowerMetrics, EnvironmentMetrics, LocalStats||
|ROUTING_APP|Routing||
|NODEINFO_APP|User|(if I remember correctly)|

Going forward we will look into making channel utilization more visible in the interface and I also agree with @mblauser that some level of caching will help with unnecessary trace routes on startup.  We would need to make sure cached data is linked with the specific connected device.

nullrouten0, 2025-02-28:
> Thanks guys, this is helpful.
> 
> If you go here, you can see who in your cluster is running MeshSense right now (larger purple icons that remain displayed on the map when you zoom out):
> 
> https://meshsense.affirmatech.com/

People sometimes turn off the reporting for one.  Also, I ran meshsense yesterday, and don't see purple icon on the map. This is showing "right now"?    We've campaigned users to stop running the software unless sitting there watching it, and turn it off when they leave.... until this is figured out

> 
> Are you 100% sure the "storms" are mostly traceroutes? Since all nodes transmit relatively frequently anyway, I'm not clear why one traceroute request would induce more TRs? When I send out a TR here, it doesn't generate more in our 32 mesh cluster.

Nodes respond in many ways when they are tickled.  The issue is this is a scripted automatic tickle of every node in the mesh that would not have occurred naturally.   Nodes respond to node-ID with more node-id...position packets result in more position packets....  and they need to respond to traceroute for every hop in the path.  Here's a typical traceroute... one innocent little trace right?   https://meshview.bayme.sh/graph/traceroute/3088035935 ....  it hit 27 nodes that responded (probably more that didn't even make it to MQTT)...  1 outbound packet was 27 responses.... EACH ONE then flooded most of the mesh (because thats how meshes work today).  Traceroute is expensive.  You've built a tool that traceroutes the entire mesh for fun.  Then it keeps going. more and more traces as "hops change".... whats the value?   in tracing every node every hour or two?  132 nodes see the 27 responses above every 2 hours is 42768 packets a day from 1 person running 1 traceroute to each node once every 2 hours... which MeshSense would sit there and happily do.    Now have 5-10 users doing this and you can see why we're concerned.  

> 
> We would really appreciate seeing actual counts of the packet types during these storms from multiple clusters. It only takes a few minutes to perform the analysis with the tools MeshSense provides. Please see above for the easy steps.

traceroute is not the dominant packet type often.  It's simply the most expensive to pass.   

Some meshes probably aren't noticing this because they were already at 40% Chanel utilization to begin with and this is just continuing to cause dropped packets and delays.  Some meshes probably have less "hop count" changes than we do.  We oscillate from 1 to 4 hops and back on simple point to point conversations happening over 60 seconds...  because there are up to 3 visible high altitude routers in view (dont tell me I need to have 1 router to support the whole Bay Area thats not going to work.).   LongFast is a dumpster fire in most places... so it's hard to notice this.  We've hand-invited people over to the MediumSlow mesh.... and enjoyed a 10-20% utilization for 132 devices owned by about 55 humans... until just 2-3 people ran mesh sense simultaneously and things went to 40%  Did the mesh survive?  of course... but is this going to get worst? absolutely... was it needed? no... traceroute is EXPENSIVE. 

Here's just 2-3 users causing hundreds of traceroute packets  in one hour 

[image]

> It still seems like there might be something more at play here than traceroutes. If you get a chance to report a packet type count, that would be extremely helpful, not only ensuring us that we are [...]

here are other port numbers from those users at that time. 

[image]

> 
> Thanks all!

N3FJP, 2025-02-28:
Excellent, thanks for all the great information.  I have a much better understanding now.

Yes, only those who have enabled the option to upload to the global map will appear as larger, purple icons.

@Soltares is in the process of deploying the new beta now as detailed in his post a short time ago.

Thanks again!

GUVWAF, 2025-02-28:
@Soltares Thanks a lot for the quick response on this. I think those are much more sensible defaults already. Being able to track the number of outgoing traceroutes is also great!

I'll close this issue now; let's see how it develops.

closed 2025-02-28 by GUVWAF

Soltares, 2025-02-28:
Version `v1.1.0-beta.2` is now live with these changes for Linux, Mac, and Windows 👍 

MeshSense checks for updates on startup and every 2 hours.  Users of the beta channel will be prompted and MeshSense will reboot as part of the update.  The reboot may trigger some initial trace routes since runtime data will be cleared, but hopefully it results in a net positive.

I have created #67 to track caching trace route data across MeshSense restarts to reduce trace route requests on startup.

Talie5in, 2025-03-01:
@Soltares I feel 15 seconds as a minimum is a bit low (personal opinion), 4 per minute from a source, that scales the rebroadcast with more nodes within an area (think 100+ node meshes and larger)

60 seconds would be for greater help, but adding scaling based on "active nodes" (like the firmware does with other packets) will help further with decreasing unhelpful (personal opinion) congestion.

Soltares, 2025-03-01:
Hi @Talie5in -- if you are referring to the `tracerouteRateLimit`, it's new minimum is 15 minutes, not seconds.  The default is 60 minutes, so a user would need to specifically lower it to 15 minutes if they wanted that rate.