Affirmatech/MeshSense #79: ACCESS_KEY not used

NickAmSee0610, 2025-03-10:
I'm starting MeshSense using: PORT=5920 ACCESS_KEY=X npm run dev

I also set the ACCESS_KEY using: export ACCESS_KEY=X

I'm accessing MeshSense by: http://192.168.178.142:5920/?ACCESS_KEY=X

But still having no connect button

Soltares, 2025-03-24:
Submitting the client-side Access_Key is not a URL parameter, but a field in the settings screen:

[image]

The access key is case-sensitive.  I tried your command above and was able to get access by inputting "X" into the front-end and clicking Apply.

```
PORT=5920 ACCESS_KEY=X npm run dev
```

NickAmSee0610, 2025-03-24:
I used this setting but no change. So I changed the code to not check the access key. Now I do have the button but I getting the following error:
Uncaught (in promise) 
Object { stack: "AxiosError@http://192.168.178.142:5922/node_modules/.vite/deps/axios.js?v=2cfdf1c6:397:18\nsettle@http://192.168.178.142:5922/node_modules/.vite/deps/axios.js?v=2cfdf1c6:1230:12\nonloadend@http://192.168.178.142:5922/node_modules/.vite/deps/axios.js?v=2cfdf1c6:1593:13\n", message: "Request failed with status code 404", name: "AxiosError", code: "ERR_BAD_REQUEST", config: {…}, request: XMLHttpRequest, response: {…}, status: 404 }
2
Failed websocket connection to /ws

Is the server variant really working?

Soltares, 2025-03-24:
My suggestion would be to check the ports configured for the API and UI.

Following [README.md](https://github.com/Affirmatech/MeshSense?tab=readme-ov-file#development-setup) the commands would be as follows:

```
cd ui
PORT=5921 npm run dev
```

And in a separate terminal

```
cd api
export DEV_UI_URL=http://localhost:5921
ACCESS_KEY=X PORT=5920 npm run dev
```

Then in a browser connect to the URL for the **API** and not the **UI**.  (The API service automatically routes front-end requests to the UI service)  API default is: http://localhost:5920

NickAmSee0610, 2025-03-25:
Thx, it's working now. Was great if this was documented in the howto

closed 2025-03-25 by NickAmSee0610