Affirmatech/MeshSense #83: [BUG] possible bug with meshsense and the client api

willp618, 2025-03-22:
meshsense version 1.1.0-beta.2 (it does it on 1.0.18 from the website download as well) 

node is heltec v3 running 2.5.23 bf958ed from the web flasher

whats happening is i was watching the logs on the serial terminal and notice that if i connect with meshsense like i normally do it will open a connection with the api then it will say after it done getting the node info and config ERROR client closed the connection 

i tried another non meshtastic progarm called meshchat  https://github.com/pablorevilla-meshtastic/MeshChat and the logs show it doesnt drop the connection 

however meshsense will continue working for abit then drop the connection requiring a reconnect 
  
attached are logs from the meshtastic flasher serial monitor one for meshchat and one for meshsense 

[meshchat logs from meshtastic flasher.txt](https://github.com/user-attachments/files/19403117/meshchat.logs.from.meshtastic.flasher.txt)

[meshsense logs from meshtastic flasher.txt](https://github.com/user-attachments/files/19403119/meshsense.logs.from.meshtastic.flasher.txt)

Soltares, 2025-03-24:
Might I ask what OS you are running?  What stands out to me in the log is that that at some point MeshSense no longer had access to any bluetooth adapters on the system:

```
[bluetooth] Unable to detect Bluetooth adapters
[bluetooth] Adapter available: false
```

willp618, 2025-03-24:
ubuntu 22.10 thinkpad x230 no bluetooth adapters as it doesn't have bluetooth i was connecting over ip

Soltares, 2025-03-24:
Thanks!  Latest I've tested here is 2.5.13.  I'll try 2.6.1 and see if I see the same behavior.

willp618, 2025-03-29:
saw a update was out so open meshsense and got it to update same node hardware same firmware of 2.6.1 and connecting over ip on thinkpad x230 it appears that the error client close issue has knock on wood been fixed and my node can connect 
however it looks like the issue with messages not getting sent properly is still present if i send a message on longfast with meshsense it is received by my other nodes however it doesnt seem to get out to the wider mesh there is a node that when not connected to meshsense i can get 100% of the time but when connected to meshsense he doesnt get my messages, we both are feeding liam cottles map via mqtt, im wondering if that causing something to go wrong as looking at the log it seems like it doesnt know what to do with the messages i have been sending 

[meshsense 1.1.0.beta.3 log message issue.txt](https://github.com/user-attachments/files/19522918/meshsense.1.1.0.beta.3.log.message.issue.txt)

jhstatewide, 2025-09-16:
> saw a update was out so open meshsense and got it to update same node hardware same firmware of 2.6.1 and connecting over ip on thinkpad x230 it appears that the error client close issue has knock [...]

Looks like some kind of bug with the firmware stripping out all punctuation.