Affirmatech/MeshSense #84: [Feature] Ability to override position for a Node

Soltares, 2025-03-24:
This was touched on briefly in #2, but a user has asked for the ability to override or hide positions of nodes that are clearly set incorrectly.

I wonder if another solution to this is to simply hide routes that are excessively long.  This gets tricky though as a node that is in an airplane or over an unobstructed waterway may be legitimately lengthy.

mblauser, 2025-03-24:
perhaps simply a manual _toggle_ ability to ignore plotting lines to specific nodes. Upon ignoring a node either:

1. Keep node position plotted on the map but remove its existing route lines and refuse further RouteRequests 
2. Remove node position and its existing route lines

Soupbones, 2025-05-06:
You pretty much wont get a node go over 400Km, maybe add it so that nodes with more than 400km range from your node are not included.

mblauser, 2025-05-22:
I wish to implement a list of positions for specific nodes to appear on the map which are: 
- not reporting position, has no approximate position calculated
- not reporting position, does have approximate position calculated
- reporting an inaccurate or undesired position
 
The map should behave differently based on a boolean override flag value (null/false, true) for each node with specified lat/long:
- if [null/false] places node at specified position which may otherwise not appear on map, map would override (or not calculate) approximate calculated position, but will NOT override position if the specified node ultimately broadcasts/reports its position.
- if [true] override both calculated and/or reported position
- if either latitude or longitude are not specified, treat this as a node placeholder and ignore the entry

Representing this data in a table:
|     id    |     num    | override |     latitude    |     longitude     |
|:---------:|:----------:|:--------:|:----------:|:------------:|
| !6eb6e0ff | 1857478811 | null     | 36.1753068 | -115.1644340 |
| !8658a541 | 2253956426 | false    | 36.3089440 | -115.3231458 |
| !6b73dc99 | 1702755122 | true     | 36.1414600 | -115.1885500 |
| !4c3569fa | 1127495113 | null     |            |              |

Should we save/load this table information via a new CSV, sqllite, JSON, or an existing storage area within the app?

Soltares, 2025-05-23:
The implementation of caching route information via `routeCache` is working well and we could use a similar approach here.

Currently routeCache is designed to be specific to the node we are connecting to.  Would the location overrides apply for all connectable devices, or the current connected device?

I can help set this up.  Perhaps have a new field in NodeDetail with the option to set and toggle the position override for a node?

mblauser, 2025-05-23:
I'm not familiar enough with use cases for switching between connected nodes. Perhaps stick to the easier implementation of specific connected-node for now?  Would an additional global flag allow for future implementation across any connected node? 

We should also ensure that when adding this, inactive nodes are not inadvertently made visible on the map when filtering by only 'active' nodes.

An appropriate name for the new field in NodeDetail might be "setPosition".  Even better would involve moving both "approximatePosition" and "setPosition" into a parent field named "mapPosition".  Here is a formatted example:

```
{
  "mapPosition": {
    "global": false,
    "approximatePosition": {
      "longitude": -115.188550039274,
      "latitude": 36.141460082658
    },
    "setPosition": {
      "longitude": -115.1885500,
      "latitude": 36.1414600,
      "override": true
    }
  }
}
```

XTR1984, 2025-10-18:
up vote for fix position feature!