Affirmatech/MeshSense #85: [BUG] Route Lines Persist Despite Node Inactivity

mblauser, 2025-03-24:
In instances where a node having an unknown _but guessed_ position becomes inactive [in my setup, typically 90 minutes], the guess-mapped node disappears as expected, but plotted route lines persist. Toggling the Inactive visibility restores the guess-mapped node represented by a white circle and question mark, at the intersection of the plotted route lines.

Expected behavior when Inactive mode visibility is off (by default), is for the plotted route lines to disappear from the map along with the guess-mapped node; then for both the guess-mapped node and plotted route lines to appear again when toggling Inactive mode visibility.

[image]

[image]

I believe this bug was introduced when adding the feature of guessing node positions.

mblauser, 2025-03-26:
Might not be limited to only guessed node positions, as I've recently observed route lines to nodes with broadcasted positions also exhibit this behavior after becoming inactive.

Soltares, 2025-04-01:
I suppose when we go and plot routes, we should exclude routes that include inactive nodes?

```ts
    ol.plotLines(
      'routes',
      nodesWithCoords
        .filter((n) => n.trace || n.hopsAway == 0)
        .map((n) => {
          let list = [myNodeCoords, ...(n.trace?.route?.map((n) => getCoordinates(n)) || []), getCoordinates(n)]
          return list.filter((coords) => !(coords[0] == 0 && coords[1] == 0))
        })
    )
```

Soltares, 2025-04-02:
Published in `v1.1.0-beta.5`

closed 2025-04-02 by Soltares