Affirmatech/MeshSense #94: MeshSense Beta 1.1.0.3 and Meshtastic 2.6.3 firmware

BazlTech, 2025-03-29:
I've been using MesshSense 1.1.0-beta.3 since it was available without issue on a heltech v3 node that was running 2.5 Meshtasticfirmware.

Today, I updated the firmware on the heltech v3 to 2.6.3.  Since then, MeshSense has had issues displaying the icons?  It now includes the channel utilization which is nice.

As you can see in the screenshot, the map and the node list have issues displaying any icons properly.

[image]

Soupbones, 2025-03-30:
I am also getting this issue, and it's not reflected on the global map. The Global map shows the nodes linking. Meshsense itself isn't showing them.  Also with MQTT selected the icon seem to shimmer. I have attached two pics. One is a screen shot of Meshsense where you can see the path with no nodes showing. The other is taken the same time and is taken from the global map. 

[image]

[image]

Soltares, 2025-03-31:
Interesting, thanks for the report.  MeshSense is currently using an external API to generate the icon.  It's possible the resource was down or unavailable.  It might be worth looking to see if it's possible to bundle the icon generation within MeshSense so it can perform this function locally.

```ts
  export function getIconURL(node: NodeInfo) {
    if (node.position?.latitudeI) {
      if (node?.position?.altitude > 2743) return `${import.meta.env.VITE_PATH || ''}/airplane.svg`
      else return `https://icongaga-api.bytedancer.workers.dev/api/genHexer?name=${node.num}`
    } else {
      return `${import.meta.env.VITE_PATH || ''}/circle-help.svg`
    }
  }
```

Soltares, 2025-04-02:
Published in `v1.1.0-beta.5`

RingoAl, 2025-04-11:
Hello.
Icons in nodes list are still external.

mblauser, 2025-05-20:
This issue existed entirely or in part by an API rate limit.  I had not yet upgraded to v1.1.0-beta.5 and received the following response when loading the icon URLs directly:

Please check back later
Error 1027
 This website has been temporarily rate limited

You cannot access this site because the owner has reached their plan limits. Check back later once traffic has gone down.

If you are owner of this website, prevent this from happening again by upgrading your plan on the [Cloudflare Workers dashboard](https://dash.cloudflare.com/?account=workers/plans).

chris-spencer, 2025-09-18:
Im running beta 1.1.0.7 and I'm still seeing this icon issue on the webUI (Screenshot attached), I think this might need re-opening? is there anything else I can give to help?

[image]