Affirmatech/MeshSense #96: Node list sorting weirdness

jwarchol, 2025-03-30:
I've noticed a few times where the node list seems to be sorted oddly where it's puts nodes with higher hop count before nodes with lower, sort of like it had two node lists, sorted them independently, then concatenated them. Just saying that's how it looks, no idea if that's happening. This is with the 1.1.0-alpha.4 but I've seen it in previous releases. In this shot I'm loading MeshSense over the local network in a Chrome browser. I do see it locally on the Windows computer that's actually running MeshSense as well.

[image]

jwarchol, 2025-03-30:
Note there's about 38 nodes above the transition point between !a756dda3 and CARD

Soupbones, 2025-03-31:
We definitely need a node sorting feature added like the Meshtastic apps have.

Soltares, 2025-03-31:
Thanks for the report!  And agreed, allowing search & sort would be ideal.  The current code is follows but can likely be cleaned up:

```
      .sort((a, b) => {
        if (a.num === $myNodeNum) return -1
        if (b.num === $myNodeNum) return 1
        if (a.hopsAway == 0 && b.hopsAway == 0) return b.snr - a.snr
        return a.hopsAway === b.hopsAway ? getNodeName(a)?.localeCompare(getNodeName(b)) : a.hopsAway - b.hopsAway
      })
```

It tries to do a few conditional-based sorts, but it's probably adding too much complexity.  I'll try to look at this sooner rather than later.

jwarchol, 2025-03-31:
The discontinuity in the sorting I'm observing seems to happen when MeshSense has a `?` for the hops. 

[image]

Here's the details on that node:

[image]

Probably a very old firmware on a node or something, but maybe we can just shove those to the end so the rest of the sorting mixes in properly?

andyshinn, 2025-06-10:
I wonder if #126 solves this? Take a look at the feature and let me know if there is anything else with sorting you are hoping to see added as a feature and I can take another look.