Affirmatech/MeshSense #97: [Feature] Display Filters for Node List

mblauser, 2025-04-03:
It might be time to consider adding display filters with the increasing amount of node data to represent:
- MQTT #56 
- active/inactive visibility #89
- include nodes lacking longName/shortName 
- additional sorting methods #96 
- nodes without positions #82 
- route line colorization options
  - by Hops #58 
  - by Signal Strength (extrapolated idea from #47 and #58) 

I thought about whether to include these within a new tab under global settings, but a lot of the filter preferences mentioned are temporary, or likely to change frequently.  Although it might make sense to persist these filter preferences.

concept mock up:

[image]

inversenebula, 2025-04-05:
This looks exactly like what I would love to see. Drowning in hundreds of nodes in my area. TIA if you go for it.

CloudHotShot, 2025-04-23:
Some additional suggestions:
- On the left-side Nodes list, provide a way to filter by full or partial node name (both short and long names)
- On the node traffic list (below the map), in addition to the node name filter, provide a way to filter by 'Type' (multi-select)
- On the node map, when clicking a node, display a popup with node details

Soupbones, 2025-05-13:
I was going to post a feature suggestion, but I'm glad to see this has been suggested. Finding nodes on Meshsense is a bit of a nightmare, and some sort of node sorting system would be a huge advantage.