Affirmatech/MeshSense #102: [FEATURE] Connection by IP - Add auto reconnect checkbox

krzotr, 2025-04-12:
I connected to the Meshtastic device via IP address and left MeshSense on overnight.
I rechecked the MeshSense events tab this morning, and the last event was from `00:44:38` instead of `07:17`. I am sure my device was working correctly all night. The left top corner was as below `Connected` and green dot - but is incorrect.

[image]

In the Log tab, I see was issue at `00:45:27:307` and the device was disconnected, as said before left top corner was `Connected`:

```
00:44:36:084 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received TELEMETRY_APP packet
00:44:36:918 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received NODEINFO_APP packet
00:44:38:088 TRACE [iMeshDevice:HttpConnection] HandleMeshPacket 📦 Received NODEINFO_APP packet
00:45:27:307 ERROR [iMeshDevice:HttpConnection] ReadFromRadio ❌ fetch failed
[meshtastic] Device Status 4
[meshtastic] Device Status 5
[electron to api] {
data: { event: 'checking-for-update', body: 'Checking for update' },
ports: []
}
(...)
[meshtastic] Calling connect 192.168.1.29
Disconnecting from device
07:16:59:597 DEBUG [iMeshDevice:HttpConnection] Constructor 🔷 HttpConnection instantiated
[bluetooth] Stop Scanning
[State] connectionStatus set connecting
[meshtastic] Connecting to Node 192.168.1.29 via IP
[meshtastic] Device Status 3
```

Is there any way to add an "auto reconnect" checkbox when connecting via IP address?
For the moment status of the connection is incorrect if the device has been disconnected, and we need to manually push the `Connect` button

Another example
- Connect device via IP address
- turn off WI-FI on your device / turn off your device
- status of connection is still `Connected`

Tested on:
- Windows 11 x64
- MeshSense 1.1.0-beta.5
- Meshtastic 2.6.4.b89355f
- Devices: `seeed_xiao_s3`, `tbeam`

hyrava, 2025-04-20:
I'd like to have that "auto reconnect" feature too. For some reason my node crashes and reboots after about 8-9h when using MeshSense but it is not big deal if we can reconnect automatically.

I am running MeshSense headless on my server with tmux and using browser on another computer to use MeshSense GUI.

 I think that crash and reboot is somehow related to MeshSense, my node does not crash if it is not connected to it. When using MeshSense,  something eats all heap memory slowly until node crashes.

Hardware and software used:
- server: Ubuntu server 22.04 LTS
- client PC: Xubuntu 22.04 LTS + Firefox
- MeshSense versions tested: 1.1.0-beta5 and 1.0.18
- Meshtastic node: Heltec V3 with firmware 2.6.0.f7afa9a
- Meshtastic node connection: Wifi

Yeraze, 2025-04-26:
Another vote here.. I'm using a StationG2 and the device keeps running, but the wifi seems a bit "unstable".  I'm getting tired of having to Disconnect/Reconnect every few hours, and then losing all the data it's collecting in the meantime.

An auto-reconnect, or some method of persisting data, would be very helpful.

formtapez, 2025-04-28:
Yeah, that feature is a must-have.
My node on the roof has not the best WiFi connection - I have to press reconnect 1-2 times every day.

TheWISPRer, 2025-05-29:
I also think that this would be a good idea, but I think that implementing this alongside conserving historical data would greatly increase its usefulness - especially with those with frequent disconnects. At this time, when the connected node disconnects and reconnects, it clears message and log history.