Affirmatech/MeshSense #105: [Feature] Write log to file or send to syslog

digi-d2, 2025-04-14:
It would be great if we could write the log to a file, or even send the logs to syslog as an option. I needed some info from the logs and the software crashed and I lost the logs, so lost all my data. 
Great program by the way!

hyrava, 2025-04-20:
Even better, I'd like to have two separate logs. One log which contains only messages and another which contains everything.

Soltares, 2025-06-02:
Appreciate the suggestions!  I received an e-mail requesting this as well.  Possibilities include:

- Syslog
- Pipe to CSV file
- Webhook
- Sqlite database

Local storage allows us to reload some history as well on connection.

SwadeDotExe, 2025-09-14:
Webhook support would be fantastic! It is my understanding that Meshtastic devices only support one TCP connection at a time, which is being used by MeshSense. In my setup, I can only connect via TCP as my node is out of Bluetooth range, meaning I am only alerted to a new message if I pull up the MeshSense client. Webhooks would allow instant notification on a new message and bring back some functionality of the original Meshtastic mobile app, all while keeping MeshSense running on the backend.