Affirmatech/MeshSense #113: Node location not always showing in Meshsense.

Soupbones, 2025-04-30:
Recently, I've noticed an issue with those nodes you see that have the white icon with the question mark. One example-A friend's node I know has a line of site to the nodes I run up on my roof that is connected to Meshsense all the time, and his node always shows up as NO location on my Meshsense map. But if you get out of Meshsense and connect the node to a phone app or the Web client for Meshtastic it shows the node in its correct position. Other nodes I own also show this node's correct position. And further more, on the Meshsense global map, the node also shows there in the correct position due to other Meshsense users seeing the node in its correct position. 
 I have attached a pic below with the offending node, which has node number 0540, but this is one of a number I see on my Meshsense map that show no location but in then MT phone app or Web client show the correct location. The node in this example 0540 and if you go and look on the Global Meshsense map in Sydney Australia you will see this nodes correct location on the western each of Sydney. 

[image]

What would be the reason that on my local Meshsense map these nodes don't show on the map in their correct location, even though in the node detail they show a GPS coordinate? And also show correctly on the global map? 
I am running Meshsense 1.1.0 - Beta 5 on Windows 11

Soltares, 2025-05-06:
Is the issue that MeshSense is displaying the approximate position of node `0540` instead of the actual location which is reported by everything else?

Here are the reports the global map has received for node `0540`:

source_name | name | Reported Longitude | Approximate Longitude | last_reported
-- | -- | -- | -- | --
000c | 0540 | 150.6289405 | 150.70536942844683 | 1746538100377
TMCB | 0540 |   | 150.68267525826906 | 1746520673637
FRTs | 0540 |   |   | 1746520681061

Since there is one report that is not approximate, it is not showing as a ❔ on global map.

Can you confirm that in node `0540` in your MeshSense instance shows a location for both the `position` and `approximatePosition` keys in Node Detail?

Soupbones, 2025-05-06:
I have asked Mike, the owner of the node, to hop on here as well and comment, but I am not sure why this node is reporting different locations, as it is bolted to a large tree and cannot move.  Why doesn't Meshsense keep the last known positive location on the map marked until it receives a different position? I know this node is not only bolted to a tree but also has a fixed location set into it. The node shows its correct position on the global Meshsense map and shows its correct position on the Liam Cottle Map as well as on the app of a few of my Meshtastic devices. I can't see why Meshsense at my location is the only one being confused by its location. Is it possible to have Meshsense application read data from the global map on a nodes position and compare the app data with the global maps data and determine which one is the more accurate?

Soupbones, 2025-05-06:
So I did a DB reset on my node, I run Meshsense yesterday and today, with the Bluetooth issues as discussed in the other thread, I have done a full delete and reinstall of Meshsense, and again, it is still showing node 0540 like in the picture attached. If you look at the global map, it is right near two other Meshsense nodes, both also feeding their correct position to the global map.  My question is also, If I am feeding node data to the global map as well, does it take the reading from the other Meshsense users and show the correct position on the global map and throw out the unknown data my Meshsense seems to be showing? and is there a way that Meshsense can look at this and send the correct position back to my application to show its correct location?

[image]

Soltares, 2025-06-02:
Interesting idea to pull data from the global map.  I would worry this would defeat the purpose somewhat of trying to build a self-sufficient mesh that doesn't rely on the web, but worth thinking about.

If I recall correctly, when multiple sources send data about a specific node, the global map will prioritize the most recent information.  I _believe_ it will also prioritize the older `position` entry if the more recent report only has `approximatePosition`.  I'd want to double-check but that seems to be the case based on what you are observing.