Affirmatech/MeshSense #115: BT connection fails

habr05, 2025-05-02:
Trying to connect via BT _(Windows 11, MeshSense 1.0.18)_ to T1000-E _(f/w 2.5.18)_ and connection gets stuck on "Configuring". The log shows multiple instances of:

```
07:11:07:223 ERROR [iMeshDevice:HttpConnection] ReadFromRadio ❌ Cannot read properties of undefined (reading 'buffer')
Error sending packet 352352840 Error: Write failed
```
 Any quick fixes?

Soltares, 2025-05-06:
When I've seen "Write failed" that seems to generally indicate that the device is not paired properly.  You can try removing and re-adding the device in Windows Bluetooth settings.  https://affirmatech.com/meshsense/bluetooth

habr05, 2025-05-25:
The problem is that this error occurs during the the pairing process. The pairing never succeeds/completes.