Affirmatech/MeshSense #116: Suggestion/question: Why not let Meshsense check if "meshtastic.local" resolves and works?

sanderjo, 2025-05-04:
At a fresh start, meshsense says "If your device is on the network, enter it's IP address in the Device IP field and click Connect".

Instead of manually finding out and filling out the IP address of your meshtastic, using `meshtastic.local` works. So, why not let meshsense try that automatically?

Note: tested on Linux, but I think it works too on Windows and MacOS.

FWIW: I search meshtastic.local in issues, and only found https://github.com/Affirmatech/MeshSense/issues/75 . Nothing in source code.

[image]

[image]

```
$ ping meshtastic.local
PING meshtastic.local (192.168.1.234) 56(84) bytes of data.
64 bytes from esp32s3-3C98DC (192.168.1.234): icmp_seq=1 ttl=64 time=20.6 ms
64 bytes from esp32s3-3C98DC (192.168.1.234): icmp_seq=2 ttl=64 time=63.3 ms
64 bytes from esp32s3-3C98DC (192.168.1.234): icmp_seq=3 ttl=64 time=42.0 ms
64 bytes from esp32s3-3C98DC (192.168.1.234): icmp_seq=4 ttl=64 time=16.3 ms
```

Soltares, 2025-05-06:
I like the suggestion!  For some reason in testing my device didn't register itself to DNS, but it could be something oddly configured on my end.

sanderjo, 2025-05-06:
I think it should be implemented fail-safe: meshsense to _try_ meshtastic.local.

- If it works: Good. Done
- If it doesn't work: let user fill out IP address

I only use Ubuntu Linux, and there it works.

I tried on my company Windows laptop, and ... no resolving. Not great. Might be because company Windows has turned off Network Discovery? The company Windows laptop also doesn't see other mDNS / Bonjour / zeroconf devices, or any other network stuff. So that might be the explanation, and thus not a good testcase.

andyshinn, 2025-06-10:
If something like this is implemented I think it should be a setting and probably default to off. For example: I am using MeshSense over BLE but I _do_ have another ESP32 node that resolves to `meshtastic.local` and I would prefer that it did not automatically try to connect to that node on first launch.