Affirmatech/MeshSense #117: https://affirmatech.com/meshsense garbled with Chrome, OK with Firefox

sanderjo, 2025-05-04:
https://affirmatech.com/meshsense on Chrome (also in incognito mode) is weird:

[image]

I have to use Firefox to get a clean page:

[image]

sanderjo, 2025-05-06:
Ah, yes: Chrome on Windows is OK for me too.

So problem is with Chrome on Ubuntu Linux? I'll check a few more Ubuntu devices. And my Chromebook.

Thanks.

Soltares, 2025-05-06:
Thanks for the report.  The webpage is dynamically fetching the current versions published and updating the UI accordingly and I wonder if somehow the browser is not properly rendering the update on your Ubuntu.

Here are my Linux tests on a Mint 21 VM and my host Arch Linux:

[image]

sanderjo, 2025-05-07:
I cannot reproduce on another Ubuntu, so closing this issue. Sorry.

[image]

closed 2025-05-07 by sanderjo