Affirmatech/MeshSense #120: Unable to connect to nodes in beta6

itxpress, 2025-05-06:
Have been running beta5 for a while without issue. Only quirk is if it's left running more than 12 hours or so, the network connection on the node dies and it has to be restarted to reconnect. But we've also been having weird quirks with WiFi in the 2.6.x versions without MeshSense connected, so not sure where the issue is there. 

But when we upgraded to beta6, we cannot connect to nodes at all. Click the Connect button and nothing happens. Tried multiple nodes (TBeam and a Vis E290) Beta 5 connects to both without issue. beta6 will not. As a standard user, the screen doesn't change. Run as an admin user, when you click connect it'll say connecting, then says disconnected, then flips back to reconnecting. But never does. Can ping the nodes fine. 

This is on WIn 11 Build 24H2

Happy to gather whatever info I can to help. Also - is there any place to manually download and install the betas to roll back to a previous beta?

Soltares, 2025-05-06:
Thanks for the report!  I don't seem to be able to replicate the issue.  Would you mind sharing the contents of Settings > Log when you try to connect?

[image]

The only change in beta 6 that I could think of that would impact connect (no UI update on click) is the additional API security for some reason mistakenly blocking your access.

Unfortunately the only rollback set up right now is back to the latest production release by unchecking the Beta checkbox in settings, but the beta is a significant leap ahead of the latest production release at the moment.

Soupbones, 2025-05-06:
I just had the same issue after doing an update via the app. It would scan and show up Bluetooth devices, but clicking on them would not connect. This is also on Win11. I had to uninstall the app, reboot the PC, and reinstall it from the Affirmatech website. Then run the fresh install. Seems to be going OK now. I like how if you click the trace route Icon, the little lightning Bolt flashes red till it is done a trace route. Does this queue up, say if you click five more?

itxpress, 2025-05-06:
@Soupbones - I was just about to do the same thing. Rollback to the release then jump up to the beta6. I'll let you all know how it goes. My logs show the IP search as I type then just bluetooth scanning and detection:

`[WSS] ::1:62128 { name: 'address', action: 'set', args: [ '192.168.5' ] }
[State] address set 192.168.5
[WSS] ::1:62128 { name: 'address', action: 'set', args: [ '192.168.5.' ] }
[State] address set 192.168.5.
[WSS] ::1:62128 { name: 'address', action: 'set', args: [ '192.168.5.1' ] }
[State] address set 192.168.5.1
[WSS] ::1:62128 { name: 'address', action: 'set', args: [ '192.168.5.14' ] }
[State] address set 192.168.5.14
[WSS] ::1:62128 { name: 'address', action: 'set', args: [ '192.168.5.143' ] }
[State] address set 192.168.5.143
[bluetooth] Device Detected | d8:3b:da:8c:08:85 MBV2_0884
[bluetooth] scanning...
[bluetooth] Device Detected | d8:3b:da:8c:08:85 MBV2_0884
[bluetooth] scanning...
[bluetooth] Device Detected | 16:b3:eb:35:7f:7d Unknown or Unsupported Device (16:b3:eb:35:7f:7d)
[bluetooth] scanning...
[bluetooth] Device Detected | d8:3b:da:8c:08:85 MBV2_0884`

There is no log at all when I click the Connect button.

Soupbones, 2025-05-06:
The latest version for download is Beta 6 anyway. I didn't check the logs, but the same happened here. The app scanned and showed a list of devices it detected ( now showing the neighbour's phone! ), but clicking on my node did nothing, even after giving it ten minutes to try and connect. I did a reboot, nothing, so the uninstall and reinstall was the next thing. Seems to be working fine.

Soltares, 2025-05-06:
Interesting.  I wonder if it's specifically related to the upgrade somehow.  Just to confirm @Soupbones, are you still able to use the Connect/Disconnect buttons after reinstall and upgrade to beta 6?  (I want to make sure it's not auto-connecting from a potential connection from v1.0.18).  I'm still suspicious of the increased security.  It's supposed to allow localhost requests and requests without an accessKey set, but I wonder if there is an edge case somewhere.

itxpress, 2025-05-06:
I tried to roll it back to the release then back up to the beta. No change. The weird thing is while doing this, I never got prompted for the admin credentials. Not sure hos it pulled that off. So, I uninstalled it, and initially it said I didn't have permissions to uninstall it (vs asking for the admin password) That was weird. But I tried again and it uninstalled. Then tried to install the release 1.0.18 and I get this:

[image]

Not sure if the rollback/upgrade to beta6 without prompting for Admin creds did someting weird. I haven't restarted yet - scrambling to finish a bunch of stuff by end of the day, but then will restart and try again. Checked security suite - no alerts, so that's not it. But if that doesn't work, I'll revo it out and try fresh.

Soltares, 2025-05-06:
> Update - I just looked closer at that error and it's talking about installing in AppData. Not Program Files. Is this designed to install in the user space? That would be ideal, but I suspect that is why things were acting weird and I'm getting permission errors.

Yes, that is correct.  Electron shouldn't trigger User Account Control as it's installed in user space.  I'm not sure what all Windows would do if you tried running it as administrator -- probably best to just run it as a normal user account.

Soupbones, 2025-05-06:
> Interesting. I wonder if it's specifically related to the upgrade somehow. Just to confirm [@Soupbones](https://github.com/Soupbones), are you still able to use the Connect/Disconnect buttons after [...]

So I just tested this for you. I shut down the application and restarted it. It didn't seem to want to connect to the Bluetooth at all. I shut the bluetooth off and ten seconds later turned it back on and my node came up. It connected OK. I did notice that when the application restarts and loads the map that the map is zoomed right out with my in the middle. Have to zoom back in again to see the mesh nodes on the map.

itxpress, 2025-05-06:
I must have clicked Run As Administrator out of habit. OK I've wiped out both meshsense directories from the Admin user AppData and the local user. Reinstalled 1.0.18 (it worked fine) and used Settings to jump to beta6. Same thing - clicking Connect does nothing. OK - I've got to tap out and run to get a kiddo from soccer. I'm going to restart the computer in the AM when I get back and try again. Actually I can remote into my work machine - if I have time I'll restart and see what happens

Soupbones, 2025-05-06:
Restart is what fixed it for me.

Soltares, 2025-05-06:
I'm not at my desk at the moment, but when I get a chance I'll put in some logging to see why the connect button is getting blocked and build and Alpha for you to try.  My guess is restarting the app triggers the auto-reconnect which doesn't go through the security logic.

Soupbones, 2025-05-07:
Mine just finallly asked to install Beta 6, installed.. and no Bluetooth again. :(

Soltares, 2025-05-07:
I just was able to replicate the issue by setting an Access Key.  It doesn't occur if Access Key is blank.

I published Beta 7 a few minutes ago with a hopeful fix!  🤞

Soupbones, 2025-05-07:
OK, Beta 7 seems to have resolved the Bluetooth Issue. You can pretty much close this. :)

itxpress, 2025-05-07:
Same - I definitely had an access key set, and that fixed it Didn't even restart. Just hopped to Beta7 and the Connect button works again. You can def close this - awesome work!!!