Affirmatech/MeshSense #122: Windows version never start

TenoTrash, 2025-05-20:
Hello there.

Windows 11 Home

It was working with no problems.

There are no errors, no warnings, no logs!

Nos errors reported in the event viewer.

Uninstall also not working.

Any hint ?

TenoTrash, 2025-05-21:
I got this when run from console:

C:\Users\teno\AppData\Local\Programs\meshsense>
DIRNAME C:\Users\teno\AppData\Local\Programs\meshsense\resources\app.asar\out\main
API_PATH C:\Users\teno\AppData\Local\Programs\meshsense\resources\app.asar.unpacked\resources\api\index.cjs
[electron] Arguments [
  'C:\\Users\\teno\\AppData\\Local\\Programs\\meshsense\\MeshSense.exe'
]
[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: 'C:\\Users\\teno\\AppData\\Local\\Programs\\meshsense\\resources\\app.asar.unpacked\\resources\\api',
  staticDirectory: 'C:\\Users\\teno\\AppData\\Local\\Programs\\meshsense\\resources\\app.asar.unpacked\\resources\\api\\static',
  dataDirectory: 'C:\\Users\\teno\\AppData\\Local\\meshsense\\Data'
}
API PROCESS EXITED! 1

Vasili-Sk, 2025-09-01:
same problem i guess:
`[electron] Checking for updates on channel null
Checking for update
{
  programDirectory: 'C:\\Users\\User\\AppData\\Local\\Programs\\meshsense\\resources\\app.asar.unpacked\\resources\\api',
  staticDirectory: 'C:\\Users\\User\\AppData\\Local\\Programs\\meshsense\\resources\\app.asar.unpacked\\resources\\api\\static',
  dataDirectory: 'C:\\Users\\User\\AppData\\Local\\meshsense\\Data'
}
[bluetooth] Begin Scanning
[bluetooth] Adapter available: true
[bluetooth] scanning...
[api to electron] { event: 'setUpdateChannel', body: undefined }
[electron] Set update channel undefined
[news] Checking for news
Server listening { address: '::', family: 'IPv6', port: 5920 }
CREATING WINDOW
[electron to api] { data: { event: 'version', body: '1.0.18' }, ports: [] }
[State] version set 1.0.18
[electron to api] {
  data: { event: 'checking-for-update', body: 'Checking for update' },
  ports: []
}
[WSS] Connection from ::1:49175
Update for version 1.0.18 is not available (latest version: 1.0.18, downgrade is allowed).
[electron to api] {
  data: {
    event: 'update-not-available',
    body: {
      version: '1.0.18',
      files: [Array],
      path: 'meshsense-x64.exe',
      sha512: 'eP4+uTydfv0snObT5u1khy/B+cnXaOuXXLvLpvSR/unqMoBPpRYUh5qjQuLrd+3m8dwj/fiaK799ajos/UJEuQ==',
      releaseDate: '2025-01-17T16:26:30.319Z'
    }
  },
  ports: []
}
API PROCESS EXITED! 18446744072635810000`