Affirmatech/MeshSense #135: [Feature] Make the map sharable

hassos246, 2025-06-14:
I would like to be able to zoom in to a location on the map, copy the url and share it. But currently it's always https://meshsense.affirmatech.com in the browser url field.

RealBenAnderson, 2025-07-26:
https://meshsense.affirmatech.com/?centerNode=LINC

Soltares, 2025-07-26:
I added the centerNode URL parameter today and should accept either a short name or decimal node ID as a value.  Hopefully this gets you closer to what you need!

sbeardPA, 2025-07-26:
You're awesome !! That is exactly what I was looking for. Any way to then embed that view like a normal Google map ? We are looking to add it to our local mesh website

FiberMoose, 2025-09-06:
Hello, 
This feature works.. but at an unreasonable performance hit (1-3 minutes to load, if at all, in an iframe on a PC with Chrome, 32gb memory and 500mbps internet).
Would it be feasable at all to implement a share coordinate and elevation function, such as with google maps and OpenStreetMaps iframes?

Example:
https://www.openstreetmap.org/?#map=11/30.2579/-97.7563 
11 = the zoom level, increasing a step the closer you get.
30.2579/-97.7563 = GPS coordinates

sbeardPA, 2025-09-06:
I haven’t seen this behavior and have had the map embedded in our meshes
website svmesh.net since the day the feature was introduced. Zooms right to
the pinned coordinates and functions fine on all sorts of devices . Memory
utilization for the browser tab is on par with other sites.