Affirmatech/MeshSense #141: [Feature] Sent message acknowledgement

Soltares, 2025-06-24:
A user wrote in requesting the ability to see when a message has been acknowledged by the receiving station (as is available in the Meshtastic app)

wlockwood, 2025-08-19:
It would be great to use the icons from the Android app or something that correlates to the various states. Yellow for "heard retransmit" / "acked by another node" could be problematic when there will never be a destination ack, as in channel messages.
I would argue that if you're using colors, green should indicate that something is complete, whatever that means for that packet type... which is a bit problematic since some stuff is done as soon as it's retransmitted and other stuff needs to wait for an ack.

andyshinn, 2025-08-19:
I think that lines up mostly with what I was thinking. The gray "pending" in my example would cover both "Pending transmission" and "Cloud" in that list. I don't know that we actually have a state for "Pending transmission" or if that is helpful to differentiate.

If the icons are helpful then maybe it would make sense to have a setting to enable icons for states so that it could be either colored dots or colored icons. Not sure if this would work visually but it would help cover this request.

andyshinn, 2025-08-19:
The progress bar is a neat idea as well! I have a branch for this in progress I just need to find some more time. Also learning a bit more about how the packets actually work for acks. They are not that intuitive at first try.