Affirmatech/MeshSense #144: [Feature] Connectivity via Serial instead of BLE/WiFi

Codgedodger, 2025-07-22:
Feature request to allow direct connection to a node via usb serial attachment. 

Current setup is a raspberry pi zero 2 with a cellular modem connection for remote over the air configuration changes and firmware updates. I’d love to use Meshsense but can’t reliably use BLE. I have direct connection to the radio via serial to do typical Meshtastic cli commands.

JFRHorton, 2025-08-14:
Seconded. BLE is unreliable for remote/unattended nodes, and it'd be far easier to rely on the host computer (like a Raspberry Pi) to handle power, config, and monitoring all over a single cable.

zedn4c3r, 2025-08-24:
Thirded.  Would make Meshsense useable with the hacker gadgets expansion board for uconsole.

Dylan111111, 2025-08-25:
This feature would be nice!

BurdMahn, 2025-08-28:
Contact uses serial and it works so nicely

shorage, 2025-09-04:
Would be nice, be able to get a pi in the air with it, flash, config, meshsense. All done remotely.